Saur0o0n/PIDKiln #1: LCD and SSR questions

rocketbob, 2019-10-18:
Thanks for putting this nice project up on github!

Question: as I understand the maximum gate current on an IO pin on the ESP32 is 12mA.  Is that enough to drive the SSR?  In my case I have two heating elements in my kiln and will have to drive them from separate pins because I don't think the ESP can source enough current to drive both from one pin.

Saur0o0n, 2019-10-20:
As I understand, fully safe current from IO pin is 12mA, but it can get higher as 50mA - this is just to make this information fully true.
I also have two heating elements, but I control them with single 40A SSR, like you can see it here:
[image] - more [here](https://adrian.siemieniak.net/portal/pidkiln-0-9-still-second-update/).
But if you would need two separate SSR - it's going to be extremely easy to do - just let me know.

Saur0o0n, 2019-10-21:
Ok, so I've tested mine SSR and it draws max 8,73mA (at least this is how much my multimeter can measure in MAX mode - I don't know it's frequency). I've looked also at some different specs for Fotek SSRs - and some says MAX 12mA, some says 7mA at 12V, and also I definitely don't have original Fotek :).
So long story short - one SSR is safe, two SSRs well.. should be safe, but better is to:

- drive two(more) heaters with single SSR (if it's in range of SSR specs - for Chinese 2x the specs ;) ). 
- add an transistor (like 2N3904) and resistor (330R / 330 Ohm should limit current to 10mA) to drive two(more) SSRs
- use additional GPIO for another SSR (I'll add it)

Saur0o0n, 2019-10-22:
I have another idea - quite easy to implement :). You could use logic level converter module like [this](https://www.aliexpress.com/item/32916020205.html?spm=a2g0s.9042311.0.0.27424c4dfihkjk).
[image]

This way you can have up to 8 SSRs - driven with single GPIO (with this chip/board). Just connect one side VA to 3,3V and ESP32 pin 19 to all A0-A8 gates. Then on the other side connect +5V to VB and B0-B8 pins to SSRs you have. This chip can (according to specs) provide up to 50mA of current on single pin.