Saur0o0n/PIDKiln #2: SPI error

Adrian8x, 2020-04-14:
Hi,

Do you have any idea why I get this error?

C:\PIDKiln\PIDKiln_addons.ino: In function 'void Setup_Addons()':

PIDKiln_addons:233:32: error: no matching function for call to 'MAX31855::begin(SPIClass*&)'

   ThermocoupleA.begin(ESP32_SPI);
[image]

Saur0o0n, 2020-04-14:
Hi,

Did you clone MAX31855 library from my repo? It looks like it's from original branch - without multiple SPI feature.

Adrian8x, 2020-04-14:
Sorry, For the beginning I used original branch. Now I get your library and the sketch is compiled successfully.

Thank you for your help and for this great library!

Adrian8x, 2020-04-14:
23:04:08.828 -> WiFi mode: 1, Retry count: 9, is wifi enabled: 4
23:04:08.874 -> [NET] Y:2020 [NET] M:03 [NET] D:20 [NET] H:11 [NET] m:00 [NET] s:00
23:04:08.968 -> [NET] Creating WiFi Access Point (AP)192.168.4.1
23:04:09.109 -> [PRG] Loading directory...
23:04:09.156 -> [PRG]	counted 6 files
23:04:09.249 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
23:04:09.249 -> Core 1 register dump:
23:04:09.249 -> PC      : 0x400013df  PS      : 0x00060830  A0      : 0x800d97f1  A1      : 0x3ffb1cb0  
23:04:09.249 -> A2      : 0x00000000  A3      : 0x3ffb1d56  A4      : 0x000000ff  A5      : 0x0000ff00  
23:04:09.249 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000070  A9      : 0x00000069  
23:04:09.249 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x7840332b  A13     : 0x3ffc2588  
23:04:09.249 -> A14     : 0x00000001  A15     : 0x3f4058f3  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
23:04:09.296 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  
23:04:09.296 -> 
23:04:09.296 -> Backtrace: 0x400013df:0x3ffb1cb0 0x400d97ee:0x3ffb1cc0 0x400d9a28:0x3ffb1d90 0x400da2dc:0x3ffb1f60 0x400eaa13:0x3ffb1fb0 0x4008909d:0x3ffb1fd0
23:04:09.296 -> 
23:04:09.296 -> Rebooting...
23:04:09.296 -> ets Jun  8 2016 00:22:57
23:04:09.296 -> 
23:04:09.296 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
23:04:09.296 -> configsip: 0, SPIWP:0xee

This is the result :((

Saur0o0n, 2020-04-14:
It looks like there is some file indexing error - next step should be like this
```[NET] Y:2020 [NET] M:02 [NET] D:20 [NET] H:11 [NET] m:00 [NET] s:00
[NET] Creating WiFi Access Point (AP)192.168.10.1
[PRG] Loading directory...
[PRG]	counted 7 files
[PRG] FName: program1.txt	 FSize:491	Sel:0
[PRG] FName: vfail1.txt	 FSize:246	Sel:0
[PRG] FName: test_up_down.txt	 FSize:270	Sel:0
```
You can decode this backtrace with Esp Exception Decoder (addon for Arduino IDE).
But I think there is something wrong with file uploaded to SPIFFS.

Adrian8x, 2020-04-14:
I get this message:

PC: 0x400013df
EXCVADDR: 0x00000000

Decoding stack results
0x400d95fe: Load_programs_dir() at C:\PIDKiln/PIDKiln_program.ino line 152
0x400d9838: Generate_INDEX() at C:\PIDKiln/PIDKiln_http.ino line 200
0x400da0ec: setup() at C:\PIDKiln/PIDKiln.ino line 154
0x400ea573: loopTask(void*) at C:\Users\Ellusia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14
0x40088cd1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Adrian8x, 2020-04-14:
I uploaded he sketch with OTA upload tool you provided and I get the same result. This is strange.

I don't understand why I have only 6 files uploaded and you have 7.

Saur0o0n, 2020-04-14:
There are 6 programs, but after first run it will generate index - so it's the seventh file, that you actually can't see as a program (it's html).
It fails on strcpy. Can you upload by OTA both program and spiffs? Perhaps there is something else in file naming on Windows (and file length) - I've been developing it on Linux.
What ESP do you have? It has to be wroover - if it's wroom - there is no psram.

Adrian8x, 2020-04-15:
I uploaded with OTA the Prog/SPIFFS and I have the same error. 
I use ESP32-WROOM-32D. That's the problem if you use PSRAM.
There is a possibilities to work with wroom? Otherwise I will buy wrover.

Saur0o0n, 2020-04-15:
You would have to replace all the memory functions from ps_XX to without ps_ - so ps_malloc to just malloc etc. There is only few of them. I'm not sure if this would be all - but it should.
```
PIDKiln_LCD.ino:      Program_run_desc=(char *)ps_malloc((strlen(msg)+1)*sizeof(char));
PIDKiln_LCD.ino:      Program_run_name=(char *)ps_malloc((strlen(msg)+1)*sizeof(char));
PIDKiln_http.ino:          Errors=(char *)ps_realloc(Errors,(strlen(Errors)+p->name().length()+3)*sizeof(char));
PIDKiln_http.ino:  screenshot=(char *)ps_malloc(SCREEN_W*SCREEN_H*2*sizeof(char)+1);
PIDKiln_logs.ino:  Logs_DIR=(DIRECTORY*)ps_malloc(sizeof(DIRECTORY)*count);
PIDKiln_program.ino:  Programs_DIR=(DIRECTORY*)ps_malloc(sizeof(DIRECTORY)*count);
PIDKiln_program.ino:      Program_run=(PROGRAM *)ps_realloc(Program_run,sizeof(PROGRAM)*Program_run_size);
PIDKiln_program.ino:  Program_run=(PROGRAM *)ps_malloc(sizeof(PROGRAM)*Program_size);
PIDKiln_program.ino:  Program_run_desc=(char *)ps_malloc((Program_desc.length()+1)*sizeof(char));
PIDKiln_program.ino:  Program_run_name=(char *)ps_malloc((Program_name.length()+1)*sizeof(char));
```

seesoe, 2020-06-22:
I used the ESP32-WROOM with no PSRAM and had the same issues. I changes all the lines to not have `sp_` and all is working now.

Saur0o0n, 2020-06-22:
Good to know :) I had Wroover, so I've used psram - probably overkill :)

kendrick90, 2020-09-18:
You should post this to the main page because wroom boards are so much more common other users are likely to have those.

Saur0o0n, 2020-09-19:
> You should post this to the main page because wroom boards are so much more common other users are likely to have those.

Well, it's there - many times. In component list, BOM and in configuration description there is even information why Wrover not Wroom.

kendrick90, 2020-09-19:
Is there a reason other than because it's what you had laying around? You
yourself said it's probably overkill to use the psram. I meant you should
add the instructions for how to compile for wroom. So people can get
started with what they have laying around.

closed 2021-12-03 by Saur0o0n