Saur0o0n/PIDKiln #3: Cone Programs

seesoe, 2020-07-13:
@Saur0o0n I was wondering how your program ramps look in relation to a degrees per hour firing schedule?

Here's an example schedule from a Skutt electronic controller for a slow 04 bisque fire.

```
Segment 1: 80 degrees / hour to 250 degrees F (27 c and 121 c)
Segment 2: 250 degrees / hour to 1000 degrees F
Segment 3: 150 degrees / hour to 1300 degrees F
Segment 4: 180 degrees / hour to 1685 degrees F
Segment 5: 80 degrees / hour to 1928 degrees F
```

Would segment 1 `27 c per hour to 121 c` be converted to
`121 / 27 = 4.48`
`60 / (121 / 27) = 13.38`

```
27:13:0
54:13:0
81:13:0
108:13:0
121:8:0
```

Saur0o0n, 2020-07-14:
Have you seen doc https://github.com/Saur0o0n/pidkiln/wiki/Program-structure ?
Your first target is to acquire 121C with increase ratio of 21C per hour. If that's correct, this should look like this:
Assuming room/start temperature 21C, you need to go up 100C. So 100/21=4,76 hours. 4,76*60 (to convert to minutes) = 286 (rounded up). This gives first line of the program:
121:286:0

Assuming you are not going to dwell at the end (so there is 0 minutes).

seesoe, 2020-07-14:
Yes I read all the docs, but does not mention best example for work time per hour.

Something like this? Seems like a really long 04 bisque profile.
```
# Candle then slow bisque at 04

# Candle for 10 hours
#83:15:585

# Segment 1: 27 degrees / hour to 121 degrees C
# ((121 - 83) / 27) * 60
121:85:0 # Starting at 83

# Segment 2: 121 degrees / hour to 538 degrees C
# ((538 - 121) / 121) * 60
538:207:0 # Starting at 121

# Segment 3: 66 degrees / hour to 705 degrees C
# ((705 - 538) / 66) * 60
705:152:0 # Starting at 538

# Segment 4: 83 degrees / hour to 919 degrees C
# ((919 - 705) / 83) * 60
919:155:0 # Starting at 705

# Segment 5: 27 degrees / hour to 1054 degrees C
# ((1054 - 919) / 27) * 60
1054:300:15 # Starting at 919
# Soak for 15 min
```

Saur0o0n, 2020-07-14:
Your program will look like this:
[image]

closed 2020-07-15 by seesoe