Saur0o0n/PIDKiln #4: Error log id values

seesoe, 2020-07-15:
Where can I find the values for the error log? I have a program that aborted twice around the same spot and I can't seem to figure out why. The kiln wasn't moved or touched in both cases.

```
Program name: 1.txt
Program desc: Slow bisque at 04
Started at: 2020-07-14 05:05:22
Possible end at: 2020-07-14 10:25:22
PID values. Kp:20.00 Ki:0.20 Kd:0.10
Start temperature: 41.8C
CSV filename: 2020-07-14 10:25:22
-=-=-= Starting program =-=-=-=-
Program ended at: 2020-07-14 08:09:20
End temperature: 690.7C
Used power: 1.8 W/h
Program aborted with error: 5
```
```
Program name: 1.txt
Program desc: Slow bisque at 04
Started at: 2020-07-15 03:52:53
Possible end at: 2020-07-15 09:12:53
PID values. Kp:20.00 Ki:0.20 Kd:0.10
Start temperature: 68.5C
CSV filename: 2020-07-15 09:12:53
-=-=-= Starting program =-=-=-=-
Program ended at: 2020-07-15 06:24:43
End temperature: 601.6C
Used power: 1.8 W/h
Program aborted with error: 5
```

Saur0o0n, 2020-07-15:
In PIDKiln.h you have specified all errors (counting start with 0)
```/* 
**  Program errors:
*/
typedef enum {
  PR_ERR_FILE_LOAD,       // failed to load file
  PR_ERR_TOO_LONG_LINE,   // program line too long (there is error probably in the line - it should be max. 1111:1111:1111 - so 14 chars, if there where more PIDKiln will throw error without checking why
  PR_ERR_BAD_CHAR,        // not allowed character in program (only allowed characters are numbers and separator ":")
  PR_ERR_TOO_HOT,         // exceeded max temperature defined in MAX_Temp
  PR_ERR_TOO_COLD,        // temperature redout below MIN_Temp
  PR_ERR_MAX31A_INT_ERR,   // failed to read MAX31855 internal temperature on kiln
  PR_ERR_MAX31A_KPROBE,    // failed to read K-probe temperature on kiln
  PR_ERR_MAX31B_INT_ERR,   // failed to read MAX31855 internal temperature on case
  PR_ERR_MAX31B_KPROBE,    // failed to read K-probe temperature on case
  PR_ERR_USER_ABORT,      // user aborted
  PR_ERR_end
} PROGRAM_ERROR_STATE;
```

So error 5 is going to be "PR_ERR_MAX31A_INT_ERR,   // failed to read MAX31855 internal temperature on kiln" - this is not exact error description, since I've changed MAX library (this is for previous one - if we are using PIDKiln 1.1). The exact error you will find in Serial debug or Syslog. This should be one of those:
```
    switch (ThermocoupleA.detectThermocouple())
    {
      case MAX31855_THERMOCOUPLE_SHORT_TO_VCC:
        DBG dbgLog(LOG_ERR,"[ADDONS] ThermocoupleA short to VCC\n");
        break;
      case MAX31855_THERMOCOUPLE_SHORT_TO_GND:
        DBG dbgLog(LOG_ERR,"[ADDONS] ThermocoupleA short to GND\n");
        break;
      case MAX31855_THERMOCOUPLE_NOT_CONNECTED:
        DBG dbgLog(LOG_ERR,"[ADDONS] ThermocoupleA not connected\n");
        break;
      default:
        DBG dbgLog(LOG_ERR,"[ADDONS] ThermocoupleA unknown error, check spi cable\n");
        break;
    }
```

seesoe, 2020-07-15:
Hmm, I tried to setup syslog but I couldn’t get my server to read anything.
I checked udp traffic on the server and I see the esp sending the data to
the server but the syslog server wasn’t able to see it and or parse the
data. What could cause this error to happen twice after a similar extended
period of time each run?

Saur0o0n, 2020-07-15:
The most probable cause are some kind of interferences - check if thermocouple cable shield is grounded, check if there is no power starving for ESP32 (I had such a case - everything was working, but thermocouple failed at some point in time - it was because power supply was too weak when used longer).

seesoe, 2020-07-16:
I changed around the mains input to be much further away from the circuits and grounded the thermocouple shielding. Still getting error 5. The 5v power supply is 2 amp. I wrote a more simple program to try and test things out faster `700:20:0` and keep getting error 5 around 580-680. I'll have to borrow a laptop to connect the usb cable for serial monitor since syslog isn't broadcasting for some reason.

seesoe, 2020-07-17:
Heres the serial output I managed to get.

```
21:59:19.605 -> [LOG] Writing to log file:2020-07-17 07:55:58,663,44,142
21:59:19.605 -> [PRG] Curr temp: 663, Set_temp: 669, Incr: 0.05 Pid_out:2419.21 Step: 8
21:59:19.605 -> [ADDONS] Temperature sensor A readout: Internal temp = 34.7 	 Last temp = 664.0 	 Average kiln temp = 663.1
21:59:19.605 -> [ADDONS] Temperature sensor A readout: Internal temp = 34.8 	 Last temp = 663.8 	 Average kiln temp = 663.2
21:59:19.605 -> [ADDONS] VCC is set:3300 ; RAW Power: 20666.9W, Raw current: 89.86A, Power global:21110 W/h:44582.084419
21:59:19.605 -> [ADDONS] Temperature sensor A readout: Internal temp = 34.8 	 Last temp = 663.5 	 Average kiln temp = 663.2
21:59:19.605 -> [ADDONS] Temperature sensor A readout: Internal temp = 34.8 	 Last temp = 663.0 	 Average kiln temp = 663.2
21:59:19.605 -> [ADDONS] VCC is set:3300 ; RAW Power: 9507.0W, Raw current: 41.33A, Power global:14632 W/h:44593.496816
21:59:19.605 -> [ADDONS] ThermocoupleA unknown error, check spi cable
21:59:19.605 -> [PRG] Aborting program with error: 5
21:59:19.605 -> [PRG] Ending program cleanly
21:59:19.605 -> [LOG] Cleaning logs...
```

Saur0o0n, 2020-07-17:
```
21:59:19.605 -> [ADDONS] VCC is set:3300 ; RAW Power: 9507.0W, Raw current: 41.33A, Power global:14632 W/h:44593.496816
21:59:19.605 -> [ADDONS] ThermocoupleA unknown error, check spi cable
```
Nothing meaningful there - so ESP lost connection to MAX31855.
Do you have power meter - because those power values looks a way off - 40A ???

Anyway - what is your setup then? Do you have capacitor for thermocouples connected? Have you tried to change MAX31855 (or switch kiln with case MAX for example).
I had also quite a lot of problems with MAX31855 sensors, but most of the problems solved capacitor on Kprobe and good power supply. Also not a bad idea is to ground shielded wire from thermocouple.

seesoe, 2020-07-17:
I tried switching around the max boards with no luck. Hmmm ya the values seem to be more than double, perhaps EMI as well?

The element is 3840 watt, 16 amp, 240v kanthal.

Here's a picture of my box, it got a little messy trying to figure this issue out. I have the thermocouple shielding single point connected to ac earth. I noticed my lcd mounting was earth grounding the dc circuitry's ground, so I just unmounted that for a test running now.

[image]

Saur0o0n, 2020-07-17:
The setup looks nice. Are you sure that thermocouple is not simply broken and at some point, with certain temperature (because of thermal expansion) it's not close to ground? I know, that this should be reported by MAX31855 IC - but I haven't tested if all those error reports are accurate - as MAX library says.

seesoe, 2020-07-17:
This is the probe I'm using, it simply pokes through the firebrick, no metal or chassis around it.
https://www.theceramicshop.com/product/28923/type-k-thermocouple-81-4/
https://www.theceramicshop.com/product/6629/thermocouple-mounting-block/

I'll have to inspect the tip later today when it cools down. I don't recall measuring any continuity between the terminals and ground but I will do a sanity check.

seesoe, 2020-07-17:
Probe tip looks ok. Didn't detect any lead wire to ground continuity.

[image][image]

Saur0o0n, 2020-07-18:
I'll ask again - I don't see it on picture, and you haven't answered previous question - to you have installed capacitor at the thermocouple connector, close to MAX board?

seesoe, 2020-07-20:
For some reason I assumed (bad) that the cap under the terminal blocks would be enough. I just added a 103 capacitor on both boards but still no luck same thing is happening. I also Installed a brand new thermocouple and yet, same thing is happening.

seesoe, 2020-07-20:
I was digging through the code and realized my power sensor was a 100amp/50ma version. I have the 30anp/1v version being delivered on Tuesday.

Also I flashed a second esp32 with thermocouple b and power sensing disabled and disconnected them from the circuit in hopes to eliminate any possible noise, however I still had no luck, same issue.

Saur0o0n, 2020-07-20:
So to summarize - this happens every time at the same time or temperature? (to find out what could trigger the problem) If you don't know - please create program that will be longer then failing one, but not reaches problematic temperature. And vice versa - shorter program that reaches quicker problematic temperature.
At this point I have not other idea what could be wrong. There is a topic about fake MAX31855 boards (https://forum.arduino.cc/index.php?topic=526439.0) - I'm not saying that this is the case here - but just something to consider. If you do the test with temperature/time - perhaps this will shade some light.

seesoe, 2020-07-20:
I have a 20 minute program that if I run at cold start will successfully pass 580c, however if I run it a 3rd time, it starts to error. Also if I run a typical ￼bisque profile and the kiln is heat soaked, then it will trigger at 580 c which is about 2-3 hours into the program.

I will check my chip markings and report back.

I found this interesting comment. He ￼would get error only at high temps like me so he increased sample rate per read.

https://github.com/jbruce12000/kiln-controller/issues/5#issuecomment-502414536

seesoe, 2020-07-20:
I presume it would be putting this in a loop and only only if all iterations fail, then abort?

https://github.com/Saur0o0n/PIDKiln/blob/master/PIDKiln_addons.ino#L84

Saur0o0n, 2020-07-20:
The difference is, that with raspberry example, errors where shorted to gnd/vcc - here is unknow error (except shorted to gnd/vcc). To be honest, after MAX library change I haven't check if those errors are reported correctly.
I don't believe this could be the problem (and I haven't mentioned it before) - but at this point - your thermocoupole A cable has a loop and it's located directly at relay switch with electromagnet inside, this is probably not the best solution anyway :)

Saur0o0n, 2020-07-20:
> I presume it would be putting this in a loop and only only if all iterations fail, then abort?
> 
> https://github.com/Saur0o0n/PIDKiln/blob/master/PIDKiln_addons.ino#L84

Yes, you can do the loop/while there if you wish. But that's kind of "ugly" solution.

seesoe, 2020-07-20:
Agreed, I'll see if it can at least work or if the emi is too high I guess I'll just pull it all out as a test and then redo the whole box idea.

I also found this 13th post where he suggested heavier filtering on the input voltage as well.
https://forums.adafruit.com/viewtopic.php?f=19&t=84826#wrap

seesoe, 2020-07-20:
I was able to get syslog working on a different server and I also added the 5 detect checks.

I ran `700:20:0` at cold start and the program finished successfully twice (expected) then the 3rd time, it aborted about 10 min in. The logs showed in one instance short to grd and then 4 unknown errors. and another couple instances 5 unknown errors. After the program ended the logs reported 5 unknown errors (instead of short to grd) after about 3-5 temperature reads.

Couple examples of abort and reads after.
```
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] Graph. mxx:126 mxy:59 ttime:15 mxtemp:700 scx:840 scy:8
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] Redrawing box on graph width:73.36 fullt:900 currt:524 prop:0.582222
1 - PIDKiln-ESP32 PIDKiln - - - ???[PRG] Pid_out RAW:3758.02 Pid_out:3758.02 Now-window:3873 WindowSize:5000 Prg_state:2
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[PRG] Aborting program with error: 5
1 - PIDKiln-ESP32 PIDKiln - - - ???[PRG] Ending program cleanly
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] Loading dir: Loading directory...
1 - PIDKiln-ESP32 PIDKiln - - - ???[INPUT] Button pressed 0.150000 seconds
1 - PIDKiln-ESP32 PIDKiln - - - ??? Short press. Current view 0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] Entering menu (6) display: 7) About
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] In menu we can print 5 lines, with 2px space, and char height 10
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] Start pos is 4, chosen position is 6, screen center is 2
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] x0: 0, y0: 24, w: 128, h: 12
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] Loading dir:	counted 13 files
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_084119.csv	 FSize:892	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_084119.log	 FSize:313	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_085707.csv	 FSize:472	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_085707.log	 FSize:343	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_135759.csv	 FSize:892	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_135759.log	 FSize:313	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_141541.csv	 FSize:892	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_141541.log	 FSize:313	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_143158.csv	 FSize:304	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_143158.log	 FSize:343	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_143812.csv	 FSize:528	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[LOG] FName: 200720_143812.log	 FSize:343	Sel:0
1 - PIDKiln-ESP32 PIDKiln - - - ???[INPUT] Rotate, SCR_MENU: Encoder turn: 1, Sizeof menu 8, Menu nr 6,
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] Entering menu (7) display: 8) Restart
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] In menu we can print 5 lines, with 2px space, and char height 10
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] Start pos is 5, chosen position is 7, screen center is 2
1 - PIDKiln-ESP32 PIDKiln - - - ???[HTTP] Head of logs - copying...
1 - PIDKiln-ESP32 PIDKiln - - - ???[LCD] x0: 0, y0: 24, w: 128, h: 12
1 - PIDKiln-ESP32 PIDKiln - - - ???[INPUT] Encoder rotated 1
1 - PIDKiln-ESP32 PIDKiln - - - ???[HTTP] End of log index - copying...
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 609.2 	 Average kiln temp = 606.4
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] ThermocoupleA unknown error, check spi cable
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 609.0 	 Average kiln temp = 606.6
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 608.8 	 Average kiln temp = 606.8
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 608.5 	 Average kiln temp = 607.0
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 608.2 	 Average kiln temp = 607.1
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 607.8 	 Average kiln temp = 607.2
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 607.5 	 Average kiln temp = 607.2
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 607.0 	 Average kiln temp = 607.2
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 34.0 	 Last temp = 606.5 	 Average kiln temp = 607.1
[... cut ...]
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 33.9 	 Last temp = 584.5 	 Average kiln temp = 589.2
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 33.9 	 Last temp = 584.0 	 Average kiln temp = 588.7
1 - PIDKiln-ESP32 PIDKiln - - - ???[ADDONS] Temperature sensor A readout: Internal temp = 33.9 	 Last temp = 583.5 	 Average kiln temp = 588.2
```

I will try `1000:30:00` tomorrow at cold start and see what happens.

seesoe, 2020-07-20:
**[image]**

Saur0o0n, 2020-07-20:
Error 5 will be reported with any kind of Thermocouple A error - the exact error (short gnd, vcc) can be only found in debug output like serial or syslog.
You could try to use slower clock for SPI and if this doesn't work use software SPI, those are even slower.
```
libraries/MAX31855/src$ less MAX31855.cpp
...
  MAXSPI->beginTransaction(SPISettings(500000, MSBFIRST, SPI_MODE0)); //up to 5MHz, read MSB first, SPI mode 0, see note
```
This is the line in library. Try something much slower - it should be more resilient to EMI.

seesoe, 2020-07-21:
What do I have to change to use software serial?

I changed the spi clock to `50000` and detect loop to 10, but no luck with either.

Saur0o0n, 2020-07-21:
Max library has soft version - I haven't tested it.
```
MAX31855.cpp
MAX31855.h
MAX31855soft.cpp
MAX31855soft.h
```
There is an example for this library with soft SPI
```
#include <MAX31855soft.h>

int32_t rawData = 0;

/*
  MAX31855soft(cs, so, sck)

  cs  - chip select
  so  - serial data output
  sck - serial clock input
*/

MAX31855soft myMAX31855(3, 4, 7); //for ESP8266 change to D3 (fails to BOOT/FLASH if pin LOW), D4 (fails to BOOT/FLASH if pin LOW), D7
```

Saur0o0n, 2020-08-19:
So any luck? I've just read your topic at adafruit (https://forums.adafruit.com/viewtopic.php?f=19&t=167443) - just stump on it, browsing "the net" ;)

seesoe, 2020-08-21:
@Saur0o0n I have done many things to combat the noise. Not knowing everything yet, I ended up moving to a raspberry pi and using a wall usb plug separate from the 5v PSU running off the kiln supply. Although the noise was still fairly high at higher temperatures, I was able to use the kiln. After digging into the difference in code, it seems even the 10 time check loop I tried on the ESP wasn't the best idea, and your code as is on master is too aggressive.. The python code used with the pi doesn't fail the program on error. It just keeps going until it finds a proper reading. Although this isn't the most efficient way to operate, it works.  Errors could range from from 3-15 +/-5 readings. I need to still respond back to Mike at adafruit, he has a wealth of great information posted. I haven't given up trying to combat the noise yet, however I'm pressed for time and its "working" enough for me to work on the ceramic parts that I need to prototype and make. Thanks for all the effort thus far. To be continued.

Bovbjergdk, 2021-02-21:
First of all, thanks for a very nice system which works very well on my 10kW kiln. Great work.
I did have a lot of troubles in the beginning with intermittent error 5 - termocouple short to ground.
In the end I realized that the thermocouple extension wires I purchased from Aliexpress was bad.
After changing those, all problems were gone.

Bovbjergdk, 2021-03-15:
Modification to my last comment: The system works great with bisque firing up to about 900 °C. But the other day I tried stoneware firing which needs about 1320 °C. At about 970 °C the system gets unstable and aborts with error 5 ("[ADDONS] ThermocoupleA unknown error, check spi cable\n"). Have tried all sorts of noise decoupling and also other sensors, without any luck. The same thing happens aboout 970 °C. I was wondering if anybody has been able to go to 1320 °C?

Saur0o0n, 2021-03-15:
The temperature is not (should not) be a problem (when we talk about signals) - but this can be a problem to termocouple. Have you used it before at this temperature? I've used my kiln up to ~1060c (it's max it can produce with current heating).

Bovbjergdk, 2021-03-17:
I have not used this temperature before as I have just newly got the kiln. But according the manual it should be able to go to 1350 °C. I did try another sensor, but of the same make and type. I have now ordered another one. I have also ordered another breakout board with a MAX31856 which I will try if anything else fails. Hope to be able to solve this, because the system is really great.

Saur0o0n, 2021-03-17:
I was just asking about thermocouple probe, have you used it before - in my experience, those are often exaggerated in terms of the temperature they can withstand. One thing is the thermocouple design (proper metals for K type) and the other is proper shielding, proper welding of those metals - often it simply fails.

Bovbjergdk, 2021-03-17:
That may well be the problem. I will be back, when I have tested the one on order.

Bovbjergdk, 2021-03-22:
I have now tried another sensor with the same result. According to the enclosed log, it seems that the fault only appear once. See encloses log.
Bit 16 and bit 1 goes high, indicating a short to ground. Strange though, that an unknown error is reported.
The next readings are fine again, so the abortion is triggered by this single faulty reading. I think, that a single or a perhaps 10 faulty readings could easily be ignored to solve this problem. 

By the way, shielding and capacitors at T+ and T- are all in place. 

[https://www.dropbox.com/s/ijjh8ejno2zcyt2/Error.PNG?dl=0](url)

Saur0o0n, 2021-03-22:
So we are back to my previous idea - https://github.com/Saur0o0n/PIDKiln/issues/4#issuecomment-678629714
To be honest, I thought I've already did that :)

Saur0o0n, 2021-03-22:
I've pushed changes that adds
```
# How many thermocouple A and B errors combined, we can tolerate without aborting program.
# If you have sporadic thermocouple read errors, you can increase this value. 
MAX31855_Error_Grace_Count = 0
```
But you have to edit it in data/etc/pidkiln.conf. It's not available with the web UI (and you can't write prefs with GUI - because you will loose this value). It's also not tested (I don't have testing device at hand now - I've dissembled it). If this will work, I'll finish and add it to GUI.

Bovbjergdk, 2021-03-23:
Exellent, I will try this as soon as possible :-). Hope it works, because every test sets me 200 Dkk back in electricity cost :-(

Bovbjergdk, 2021-03-23:
Maybe the "Read_errors" counter should have be zeroed after a certain number of valid readings. As far as I can see this does not happen, so in a long run the errors could add up without beeing a problem.

Saur0o0n, 2021-03-23:
Well this sounds like a reasonable idea, but it would definitely require more rewrite, since I would have to split error resetting to different sensors (it would be stupid if one working sensor would put the other indefinitely in grace error count threshold). Will give it some thought.
Can't you just heat the thermocouple - not the entire kiln (for testing purpose)? Torch lighter flame should be plenty enough to go over 1000C

Bovbjergdk, 2021-03-24:
I tryed to heat the themocouple with a Blowtorch, but that does not introduce any faults. Therefor it must be some kind of electrical noise. 
About the "Read_errors" counter: Perhaps it should be decremented by 1 each time x consecutive valid reading is achieved.
x also beeing defined in pidkiln.conf too.

Saur0o0n, 2021-03-24:
> About the "Read_errors" counter: Perhaps it should be decremented by 1 each time x consecutive valid reading is achieved.
> x also beeing defined in pidkiln.conf too.

Sure, that was also my idea - but as I wrote before, thermocoupleA (high temperature) could fail all the time, but if thermocoupleB will work (low temperature) it will keep it error count at constant. So that would require, at least, separate error count for each probe.

Bovbjergdk, 2021-03-24:
From my intense search on the net on issues with max31855 it seems that issues are only present at high temperatures. One guy claim, that air is becoming slightly conductive above 900 °C, and that causes the problem. So perhaps it is only necessary to reprogram themocoupleA and leave ThermocoupleB as it is.
I have never seen abortions from this Thermocouple, but on the other hand I have not run the system for more than some days.

Saur0o0n, 2021-03-24:
I've splited errors count for each termocouple and I've made that each proper readout reduces error count. Still - I've just check it compiles - nothing more.

Bovbjergdk, 2021-03-27:
I just had a go with new update with MAX31855_Error_Grace_Count = 10 , but it stille aborts, here at 17:43:17. The syslog does not contain anything about the error counter. My C++ knowledge is only basic, but should the message "[ADDONS] ThermocoupleA had an error but we are still below grace threshold ........." not have been logges to syslog?

2021-03-27,17:43:19,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 29.9 	 Last temp = 1039.5 	 Average kiln temp = 1039.2 |   |   |   |  
-- | -- | -- | -- | --
2021-03-27,17:43:18,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor B readout:   Internal temp = 29.6 	 Last temp = 29.8 	 Average case temp = 29.8 |   |   |   |  
2021-03-27,17:43:18,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 30.2 	 Last temp = 1039.8 	 Average kiln temp = 1039.2 |   |   |   |  
2021-03-27,17:43:17,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[HTTP] End of log index - copying... |   |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[HTTP] Head of logs - copying... |   |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[LOG] FName: 210327_160702.log	   FSize:366	Sel:0 |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[LOG] FName: 210327_160702.csv	   FSize:9574	Sel:0 |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[LOG] FName: 210327_092121.csv	   FSize:24381	Sel:0 |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[LOG] FName: 210327_092121.log	   FSize:272	Sel:0 |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Information,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[LOG] Loading dir:	counted 5 files |   |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Information,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[LOG] Loading dir: Loading directory... |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Information,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[PRG] Ending program cleanly |   |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Information,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[PRG] Aborting program with error: 5 |   |   |   |   |   |   |   |   |  
2021-03-27,17:43:17,Error,192.168.1.99,kern,1,"-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] ThermocoupleA unknown error, check   spi cable" |   |   |   |   |   |   |  
2021-03-27,17:43:16,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 30.2 	 Last temp = 1039.8 	 Average kiln temp = 1039.1 |   |   |   |  
2021-03-27,17:43:15,Information,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[PRG] Pid_out RAW:4999.89 Pid_out:4999.89   Now-window:2357 WindowSize:5000 Prg_state:2 |   |   |   |  
2021-03-27,17:43:15,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 30.2 	 Last temp = 1039.2 	 Average kiln temp = 1039.0 |   |   |   |  
2021-03-27,17:43:14,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 30.2 	 Last temp = 1039.8 	 Average kiln temp = 1039.0 |   |   |   |  
2021-03-27,17:43:13,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 30.2 	 Last temp = 1039.0 	 Average kiln temp = 1038.9 |   |   |   |  
2021-03-27,17:43:12,Information,192.168.1.99,kern,1,"-   PIDKiln-ESP32 PIDKiln - - - ï»¿[PRG] Curr temp: 1039, Set_temp: 1047, Incr:   0.02 Pid_out:5000.00 Step: 0" |   |   |   |   |   |  
2021-03-27,17:43:12,Debug,192.168.1.99,kern,1,-   PIDKiln-ESP32 PIDKiln - - - ï»¿[ADDONS] Temperature sensor A readout:   Internal temp = 30.1 	 Last temp = 1039.5 	 Average kiln temp = 1038.9 |   |   |   |

Saur0o0n, 2021-03-27:
Like I said - it's not tested :(. There was misspell error in PIDKiln.h (MAX31855_Error_Grace_Coun instead of MAX31855_Error_Grace_Count) - fixed.

To be sure you can set it in PIDKIln_prefs.ino:
```
      case PRF_ERROR_GRACE_COUNT:
        Prefs[PRF_ERROR_GRACE_COUNT].type=UINT8;
        Prefs[PRF_ERROR_GRACE_COUNT].value.uint8=0;
        break;
```

It should log the grace if hits it (PIDKiln_addons.ino):
```
    if(TempA_errors<Prefs[PRF_ERROR_GRACE_COUNT].value.uint8){
      TempA_errors++;
      DBG dbgLog(LOG_ERR,"[ADDONS] ThermocoupleA had an error but we are still below grace threshold - continue. Error %d of %d\n",TempA_errors,Prefs[PRF_ERROR_GRACE_COUNT].value.uint8);
    }else{
      ABORT_Program(PR_ERR_MAX31A_INT_ERR);
    }
```

Bovbjergdk, 2021-04-05:
I have not had the need for any high temperature burnings yet, but I have provoked a Thermocouple failure by disconnecting the probe, and the error routine is now working perfect. I will return when I have done the real test.

Saur0o0n, 2021-04-19:
I've added MAX31855_Error_Grace_Count to the preference page, and tested it briefly.

Bovbjergdk, 2021-09-12:
Even with the counter sat to 20 the system stille shuts down at about 1060 °C.

Bovbjergdk, 2021-09-12:
It would be nice to be able to set the counter in the Preference table.

Saur0o0n, 2021-09-13:
I've added it to prefs page - read my last comment. You probably just need to upgrade.

Bovbjergdk, 2021-09-14:
Sorrry, I did not see that. I have now upgraded. Yesterday I ran a burning to 1250 °C. The system aborted at 1050 °C, but I bypassed the controller and the kiln went up to the desired 1250 °C. The controller contiued to measure and plot the temperature even if it had aborted. This tells me, that a GRACE_COUNT, which was set to 20, is not enough.  Does the counter decrement or reset, when a valid measurement is achievd?

From yesterdays burning I should be able to see in the syslog, what happened. But the syslog is some sort of database, of which I am not familiar. (SYNOSYSLOGDB__ARCH.DB). How do I open this?

Kind regards
Jens

Saur0o0n, 2021-09-14:
>  Does the counter decrement or reset, when a valid measurement is achieved?

Yes, but it exactly decrement the counter, not zero it - so if the errors still occur too often it will trigger error eventually
You could change this line (for A and B thermoucouple)
```
  if(TempA_errors>0) TempA_errors--;  // Lower errors count after proper readout
```
to
```
  if(TempA_errors>0) TempA_errors=0;  // Lower errors count after proper readout
```
If you would like to zero counter after proper readout.

> From yesterdays burning I should be able to see in the syslog, what happened. But the syslog is some sort of database, of which I am not familiar. (SYNOSYSLOGDB__ARCH.DB). How do I open this?
It depends what syslog service you are using - there are many. For Linux/Unix environments it's usually simple text file.

Bovbjergdk, 2021-09-14:
OK thanks. I think that resetting to 0 is the way to go. That means, that only if X **consecutive** errors occur, the system will abort. The X being the Grace counter setting. 

By the way, I really think this system is awesome and user friendly and very professional made.

closed 2021-12-03 by Saur0o0n