Saur0o0n/PIDKiln #7: NTP

kendrick90, 2021-05-05:
If I set the GMT_OFFSET time to -25200 it does not produce the correct time. I believe this may be an int16 overflow.

Saur0o0n, 2021-05-05:
This is in range of int16 (signed number from -32768 to 32767) - but I see some errors in NTP function, it calls it as uint, not int. I'm not sure if this is the problem - but I'll push fix for it. Please check.

kendrick90, 2021-05-06:
It works thanks! I would also suggest making it be offset in hours and just multiply by 3600 in the configTime line. Thanks again. I'll be trying it out in the next week or so after I install a 240V outlet for my kiln. Another random suggestion that might not work 100% of the time but doesn't seem to break it is `#include <ESPmDNS.h>` and  add
`      if (!MDNS.begin("PIDKiln")) {
        Serial.println("Error starting mDNS");
        return;
      }`  in PIDKiln.ino 
after connecting to the network. That way you can access the interface from http://pidkiln.local

kendrick90, 2021-05-15:
I have been working on getting this $10 screen + rotary encoder + buzzer + SD card reader + RGB LEDs ["fysetc mini 12864"](https://www.amazon.com/gp/product/B085RYXYCM/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1) working to use with the PIDKiln controller. The code I have written for the screen in platformio and I have just finished porting your project to be compatible with platformio too. https://github.com/kendrick90/PIDKilnPIO I am not a great programmer but I am learning through this project. I am hoping that we can store files on the SD card instead of the PSRAM so we can use the cheaper WROOM boards but for now I will stick with the WROVER. I also plan to make a simple pcb with headers for the screen, esp, max31855, and power to plug into. I think it will be a very clean package all said and done. Thanks for creating this project for me to work with and if you are interested in collaborating you can reach me at my user name at gmail.

closed 2021-05-15 by kendrick90