Saur0o0n/PIDKiln #8: loop in error exception

gdbdzgd, 2021-07-16:
build from source or update from ota ,all loop in the error ,any one can help me?

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13416
load:0x40080400,len:3672
entry 0x400805f8
[PREFS] Loading prefs from file[PREFS]  -> For WiFi_Retry_cnt saved UINT8 item value:9 type:1
[PREFS]  -> For WiFi_Mode saved UINT8 item value:1 type:1
[PREFS]  -> For HTTP_Local_JS saved UINT8 item value:1 type:1
[PREFS]  -> For Auth_Username saved STRING item value:admin type:4
[PREFS]  -> For Auth_Password saved STRING item value:hotashell type:4
[PREFS]  -> For NTP_Server1 saved STRING item value:0.pl.pool.ntp.org type:4
[PREFS]  -> For NTP_Server2 saved STRING item value:1.pl.pool.ntp.org type:4
[PREFS]  -> For NTP_Server3 saved STRING item value:2.pl.pool.ntp.org type:4
[PREFS]  -> For GMT_Offset_sec saved INT16 item value:3600 type:3
[PREFS]  -> For Daylight_Offset_sec saved INT16 item value:3600 type:3
[PREFS]  -> For Initial_Date saved STRING item value:2020-02-20 type:4
[PREFS]  -> For Initial_Time saved STRING item value:11:00:00 type:4
[PREFS]  -> For PID_Window saved UINT16 item value:5000 type:2
[PREFS]  -> For PID_Kp saved VFLOAT item value:20.000000 type:5
[PREFS]  -> For PID_Ki saved VFLOAT item value:0.200000 type:5
[PREFS]  -> For PID_Kd saved VFLOAT item value:0.100000 type:5
[PREFS]  -> For PID_Temp_Threshold saved INT16 item value:-1 type:3
[PREFS]  -> For LOG_Window saved UINT16 item value:30 type:2
[PREFS]  -> For LOG_Files_Limit saved UINT16 item value:40 type:2
[PREFS]  -> For DBG_Serial saved UINT8 item value:1 type:1
[PREFS]  -> For DBG_Syslog saved UINT8 item value:0 type:1
[PREFS]  -> For MIN_Temperature saved UINT8 item value:10 type:1
[PREFS]  -> For MAX_Temperature saved UINT16 item value:1350 type:2
[PREFS]  -> For MAX_Housing_Temperature saved UINT16 item value:130 type:2
[PREFS]  -> For Thermal_Runaway saved UINT16 item value:0 type:2
[PREFS]  -> For Alarm_Timeout saved UINT16 item value:5 type:2
[PREFS] -=-=-= PREFS DISPLAY =-=-=-[PREFS] 1) 'WiFi_SSID' = ''	4
[PREFS] 2) 'WiFi_Password' = ''	4
[PREFS] 3) 'WiFi_Mode' = '1'	1
[PREFS] 4) 'WiFi_Retry_cnt' = '9'	1
[PREFS] 8) 'HTTP_Local_JS' = '1'	1
[PREFS] 9) 'Auth_Username' = 'admin'	4
[PREFS] 10) 'Auth_Password' = 'hotashell'	4
[PREFS] 11) 'NTP_Server1' = '0.pl.pool.ntp.org'	4
[PREFS] 12) 'NTP_Server2' = '1.pl.pool.ntp.org'	4
[PREFS] 13) 'NTP_Server3' = '2.pl.pool.ntp.org'	4
[PREFS] 14) 'GMT_Offset_sec' = '3600'	3
[PREFS] 15) 'Daylight_Offset_sec' = '3600'	3
[PREFS] 16) 'Initial_Date' = '2020-02-20'	4
[PREFS] 17) 'Initial_Time' = '11:00:00'	4
[PREFS] 18) 'PID_Window' = '5000'	2
[PREFS] 22) 'PID_POE' = '0'	1
[PREFS] 23) 'PID_Temp_Threshold' = '-1'	3
[PREFS] 24) 'LOG_Window' = '30'	2
[PREFS] 25) 'LOG_Files_Limit' = '40'	2
[PREFS] 26) 'MIN_Temperature' = '10'	1
[PREFS] 27) 'MAX_Temperature' = '1350'	2
[PREFS] 28) 'MAX_Housing_Temperature' = '130'	2
[PREFS] 29) 'Thermal_Runaway' = '0'	2
[PREFS] 30) 'Alarm_Timeout' = '5'	2
[PREFS] 31) 'DBG_Serial' = '1'	1
[PREFS] 32) 'DBG_Syslog' = '0'	1
[PREFS] 33) 'DBG_Syslog_Srv' = ''	4
[PREFS] 34) 'DBG_Syslog_Port' = '0'	2
WiFi mode: 1, Retry count: 9, is wifi enabled: 4
[NET] Y:2020 [NET] M:02 [NET] D:20 [NET] H:11 [NET] m:00 [NET] s:00
[NET] Creating WiFi Access Point (AP)192.168.10.1
[PRG] Loading directory...
[PRG]	counted 6 files
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400013df  PS      : 0x00060830  A0      : 0x800da108  A1      : 0x3ffb1cb0  
A2      : 0x00000000  A3      : 0x3ffb1d56  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000070  A9      : 0x00000069  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc2844  A13     : 0x2a357c1e  
A14     : 0x00000000  A15     : 0x00000008  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400013df:0x3ffb1cb0 0x400da105:0x3ffb1cc0 0x400da35c:0x3ffb1d90 0x400dac28:0x3ffb1f60 0x400ebbc6:0x3ffb1fb0 0x40090be1:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

```

error decode:

```
PC: 0x4008a1e3
EXCVADDR: 0x00000001

Decoding stack results
0x400dbf29: Load_programs_dir() at /home/gdzhang/PIDKiln/PIDKiln_program.ino line 151
0x400dc202: Generate_INDEX() at /home/gdzhang/PIDKiln/PIDKiln_http.ino line 201
0x400dcfc8: setup() at /home/gdzhang/PIDKiln/PIDKiln.ino line 154
0x400f0aab: loopTask(void*) at /home/gdzhang/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 35

```

Saur0o0n, 2021-07-16:
Hi,
You probably are using ESP32 Wroom instead of Wrover - and you don't have psram. Check here (https://github.com/Saur0o0n/PIDKiln/issues/2#issuecomment-613921578), it should work without it.

gdbdzgd, 2021-07-17:
@Saur0o0n tks。 i‘ll try the #2 ’s Code。

gdbdzgd, 2021-07-19:
i change the line's and i got loop error too.

```C
diff --git a/PIDKiln_LCD.ino b/PIDKiln_LCD.ino
index 7686c8a..0752277 100644
--- a/PIDKiln_LCD.ino
+++ b/PIDKiln_LCD.ino
@@ -715,11 +715,11 @@ char msg[100];
       Initialize_program_to_run();  // clear current program
       sprintf(msg,"Manually created quick program.");
       DBG dbgLog(LOG_DEBUG,"[LCD] Replacing current program in memory:%d \n",strlen(msg));
-      Program_run_desc=(char *)ps_malloc((strlen(msg)+1)*sizeof(char));
+      Program_run_desc=(char *)malloc((strlen(msg)+1)*sizeof(char));
       strcpy(Program_run_desc,msg);
       sprintf(msg,"QuickProgram");
       DBG dbgLog(LOG_DEBUG,"[LCD] Replacing current program in memory:%d \n",strlen(msg));
-      Program_run_name=(char *)ps_malloc((strlen(msg)+1)*sizeof(char));
+      Program_run_name=(char *)malloc((strlen(msg)+1)*sizeof(char));
       strcpy(Program_run_name,msg);
       Update_program_step(0, qp[0], qp[1], qp[2]);
       Program_run_state=PR_READY;
diff --git a/PIDKiln_http.ino b/PIDKiln_http.ino
index 36ccd79..0a08021 100644
--- a/PIDKiln_http.ino
+++ b/PIDKiln_http.ino
@@ -562,7 +562,7 @@ boolean save=false;
         // we have some errors add new field to error list
         if(Errors!=NULL){
           DBG dbgLog(LOG_DEBUG,"[HTTP] Realloc call of size %d\n",(strlen(Errors)+p->name().length()+3)*sizeof(char));
-          Errors=(char *)ps_realloc(Errors,(strlen(Errors)+p->name().length()+3)*sizeof(char));
+          Errors=(char *)realloc(Errors,(strlen(Errors)+p->name().length()+3)*sizeof(char));
           strcat(Errors," ");
           strcat(Errors,p->name().c_str());
           DBG dbgLog(LOG_DEBUG,"[HTTP] Errors now:%s\n",Errors);
@@ -655,7 +655,7 @@ char *screenshot;
 void out(const char *s){strcat(screenshot,s);}
 void do_screenshot(AsyncWebServerRequest *request){
 
-  screenshot=(char *)ps_malloc(SCREEN_W*SCREEN_H*2*sizeof(char)+1);
+  screenshot=(char *)malloc(SCREEN_W*SCREEN_H*2*sizeof(char)+1);
   if(screenshot==NULL){
     DBG dbgLog(LOG_ERR,"[HTTP] Failed to allocate memory for screenshot");
     request->send(500);
diff --git a/PIDKiln_logs.ino b/PIDKiln_logs.ino
index 3023e90..d85ae59 100644
--- a/PIDKiln_logs.ino
+++ b/PIDKiln_logs.ino
@@ -140,7 +140,7 @@ File dir,file;
   
   DBG dbgLog(LOG_INFO,"[LOG] Loading dir:\tcounted %d files\n",count);
   if(Logs_DIR) free(Logs_DIR);
-  Logs_DIR=(DIRECTORY*)ps_malloc(sizeof(DIRECTORY)*count);
+  Logs_DIR=(DIRECTORY*)malloc(sizeof(DIRECTORY)*count);
   Logs_DIR_size=0;
   dir.rewindDirectory();
   while((file=dir.openNextFile()) && Logs_DIR_size<=count){    // now we do acctual loading into memory
diff --git a/PIDKiln_program.ino b/PIDKiln_program.ino
index 66d1dd8..c13500b 100644
--- a/PIDKiln_program.ino
+++ b/PIDKiln_program.ino
@@ -135,7 +135,7 @@ File dir,file;
   while(dir.openNextFile()) count++;  // not the prettiest - but we count files first to do proper malloc without fragmenting memory
   DBG dbgLog(LOG_DEBUG,"[PRG]\tcounted %d files\n",count);
   if(Programs_DIR) free(Programs_DIR);
-  Programs_DIR=(DIRECTORY*)ps_malloc(sizeof(DIRECTORY)*count);
+  Programs_DIR=(DIRECTORY*)malloc(sizeof(DIRECTORY)*count);
   Programs_DIR_size=0;
   dir.rewindDirectory();
   while((file=dir.openNextFile()) && Programs_DIR_size<=count){    // now we do acctual loading into memory
@@ -186,7 +186,7 @@ void Update_program_step(uint8_t sstep, uint16_t stemp, uint16_t stime, uint16_t
[... 134 lines cut ...]
[PREFS] 9) 'Auth_Username' = 'admin'    4
[PREFS] 10) 'Auth_Password' = 'hotashell'       4
[PREFS] 11) 'NTP_Server1' = 'pool.ntp.org'      4
[PREFS] 12) 'NTP_Server2' = ''  4
[PREFS] 13) 'NTP_Server3' = ''  4
[PREFS] 14) 'GMT_Offset_sec' = '0'      3
[PREFS] 15) 'Daylight_Offset_sec' = '0' 3
[PREFS] 16) 'Initial_Date' = '2012-02-20'       4
[PREFS] 17) 'Initial_Time' = '00:00:00' 4
[PREFS] 18) 'PID_Window' = '5000'       2
[PREFS] 22) 'PID_POE' = '0'     1
[PREFS] 23) 'PID_Temp_Threshold' = '-1' 3
[PREFS] 24) 'LOG_Window' = '30' 2
[PREFS] 25) 'LOG_Files_Limit' = '40'    2
[PREFS] 26) 'MIN_Temperature' = '10'    1
[PREFS] 27) 'MAX_Temperature' = '1350'  2
[PREFS] 28) 'MAX_Housing_Temperature' = '130'   2
[PREFS] 29) 'Thermal_Runaway' = '130'   2
[PREFS] 30) 'Alarm_Timeout' = '0'       2
[PREFS] 31) 'MAX31855_Error_Grace_Count' = '0'  1
[PREFS] 32) 'DBG_Serial' = '1'  1
[PREFS] 33) 'DBG_Syslog' = '0'  1
[PREFS] 34) 'DBG_Syslog_Srv' = ''       4
[PREFS] 35) 'DBG_Syslog_Port' = '0'     2
```

gdbdzgd, 2021-07-19:
i have bus some esp32 dev board with psram , 
but's need some days.

gdbdzgd, 2021-07-19:
> i change the line's and i got loop error too.
> 
> ```c
> diff --git a/PIDKiln_LCD.ino b/PIDKiln_LCD.ino
> index 7686c8a..0752277 100644
> --- a/PIDKiln_LCD.ino
> +++ b/PIDKiln_LCD.ino
> @@ -715,11 +715,11 @@ char msg[100];
>        Initialize_program_to_run();  // clear current program
>        sprintf(msg,"Manually created quick program.");
>        DBG dbgLog(LOG_DEBUG,"[LCD] Replacing current program in memory:%d \n",strlen(msg));
> -      Program_run_desc=(char *)ps_malloc((strlen(msg)+1)*sizeof(char));
> +      Program_run_desc=(char *)malloc((strlen(msg)+1)*sizeof(char));
>        strcpy(Program_run_desc,msg);
>        sprintf(msg,"QuickProgram");
>        DBG dbgLog(LOG_DEBUG,"[LCD] Replacing current program in memory:%d \n",strlen(msg));
> -      Program_run_name=(char *)ps_malloc((strlen(msg)+1)*sizeof(char));
> +      Program_run_name=(char *)malloc((strlen(msg)+1)*sizeof(char));
>        strcpy(Program_run_name,msg);
>        Update_program_step(0, qp[0], qp[1], qp[2]);
>        Program_run_state=PR_READY;
> diff --git a/PIDKiln_http.ino b/PIDKiln_http.ino
> index 36ccd79..0a08021 100644
> --- a/PIDKiln_http.ino
> +++ b/PIDKiln_http.ino
> @@ -562,7 +562,7 @@ boolean save=false;
>          // we have some errors add new field to error list
>          if(Errors!=NULL){
>            DBG dbgLog(LOG_DEBUG,"[HTTP] Realloc call of size %d\n",(strlen(Errors)+p->name().length()+3)*sizeof(char));
> -          Errors=(char *)ps_realloc(Errors,(strlen(Errors)+p->name().length()+3)*sizeof(char));
> +          Errors=(char *)realloc(Errors,(strlen(Errors)+p->name().length()+3)*sizeof(char));
>            strcat(Errors," ");
>            strcat(Errors,p->name().c_str());
>            DBG dbgLog(LOG_DEBUG,"[HTTP] Errors now:%s\n",Errors);
> @@ -655,7 +655,7 @@ char *screenshot;
>  void out(const char *s){strcat(screenshot,s);}
>  void do_screenshot(AsyncWebServerRequest *request){
>  
> -  screenshot=(char *)ps_malloc(SCREEN_W*SCREEN_H*2*sizeof(char)+1);
> +  screenshot=(char *)malloc(SCREEN_W*SCREEN_H*2*sizeof(char)+1);
>    if(screenshot==NULL){
>      DBG dbgLog(LOG_ERR,"[HTTP] Failed to allocate memory for screenshot");
>      request->send(500);
> diff --git a/PIDKiln_logs.ino b/PIDKiln_logs.ino
> index 3023e90..d85ae59 100644
> --- a/PIDKiln_logs.ino
> +++ b/PIDKiln_logs.ino
> @@ -140,7 +140,7 @@ File dir,file;
>    
>    DBG dbgLog(LOG_INFO,"[LOG] Loading dir:\tcounted %d files\n",count);
>    if(Logs_DIR) free(Logs_DIR);
> -  Logs_DIR=(DIRECTORY*)ps_malloc(sizeof(DIRECTORY)*count);
> +  Logs_DIR=(DIRECTORY*)malloc(sizeof(DIRECTORY)*count);
>    Logs_DIR_size=0;
>    dir.rewindDirectory();
>    while((file=dir.openNextFile()) && Logs_DIR_size<=count){    // now we do acctual loading into memory
> diff --git a/PIDKiln_program.ino b/PIDKiln_program.ino
> index 66d1dd8..c13500b 100644
> --- a/PIDKiln_program.ino
> +++ b/PIDKiln_program.ino
> @@ -135,7 +135,7 @@ File dir,file;
>    while(dir.openNextFile()) count++;  // not the prettiest - but we count files first to do proper malloc without fragmenting memory
>    DBG dbgLog(LOG_DEBUG,"[PRG]\tcounted %d files\n",count);
>    if(Programs_DIR) free(Programs_DIR);
> -  Programs_DIR=(DIRECTORY*)ps_malloc(sizeof(DIRECTORY)*count);
> +  Programs_DIR=(DIRECTORY*)malloc(sizeof(DIRECTORY)*count);
>    Programs_DIR_size=0;
>    dir.rewindDirectory();
>    while((file=dir.openNextFile()) && Programs_DIR_size<=count){    // now we do acctual loading into memory
> @@ -186,7 +186,7 @@ void Update_program_step(uint8_t sstep, uint16_t stemp, uint16_t stime, uint16_t
[... 136 lines cut ...]
> [PREFS] 11) 'NTP_Server1' = 'pool.ntp.org'      4
> [PREFS] 12) 'NTP_Server2' = ''  4
> [PREFS] 13) 'NTP_Server3' = ''  4
> [PREFS] 14) 'GMT_Offset_sec' = '0'      3
> [PREFS] 15) 'Daylight_Offset_sec' = '0' 3
> [PREFS] 16) 'Initial_Date' = '2012-02-20'       4
> [PREFS] 17) 'Initial_Time' = '00:00:00' 4
> [PREFS] 18) 'PID_Window' = '5000'       2
> [PREFS] 22) 'PID_POE' = '0'     1
> [PREFS] 23) 'PID_Temp_Threshold' = '-1' 3
> [PREFS] 24) 'LOG_Window' = '30' 2
> [PREFS] 25) 'LOG_Files_Limit' = '40'    2
> [PREFS] 26) 'MIN_Temperature' = '10'    1
> [PREFS] 27) 'MAX_Temperature' = '1350'  2
> [PREFS] 28) 'MAX_Housing_Temperature' = '130'   2
> [PREFS] 29) 'Thermal_Runaway' = '130'   2
> [PREFS] 30) 'Alarm_Timeout' = '0'       2
> [PREFS] 31) 'MAX31855_Error_Grace_Count' = '0'  1
> [PREFS] 32) 'DBG_Serial' = '1'  1
> [PREFS] 33) 'DBG_Syslog' = '0'  1
> [PREFS] 34) 'DBG_Syslog_Srv' = ''       4
> [PREFS] 35) 'DBG_Syslog_Port' = '0'     2
> ```

the loop error is the cable 's problem.  the pidkiln is work!

Saur0o0n, 2021-07-19:
You have "Brownout detector was triggered" - this usually means you have some board power issues (to low, unstable during startup etc).

gdbdzgd, 2021-08-04:
@Saur0o0n thank you .the cable's problem , voltage is too low 4.6v ,when the board is running is will triger to restart.

closed 2021-08-04 by gdbdzgd