Saur0o0n/PIDKiln #9: Painfully slow web UI

stefan63, 2021-09-24:
I try to use this without any extra hardware and want to control it via http only.
But I cant get it to work, slow loading, graphics does not load att all or slooooow.
What am I doing wrong?

Saur0o0n, 2021-09-25:
What do you mean "without any extra hardware"? You did not connect any sensors, LCD etc? Check the serial output, but it may be, that software is just busy reporting errors on missing components - I don't know, haven't tested it that way.
This may be also WiFi issue (check ping), or your ESP issue - not much information to work on.

stefan63, 2021-09-25:
Yes just uploaded the code to the esp, no sensors or ssr. I plan to use it
from web only so no display.

Saur0o0n, 2022-01-28:
Can you check latest commit? I haven't done anything with http, but LCD display was crippled and some internal tasks where also waiting much longer then should (fixed with program loop priority).