Saur0o0n/PIDKiln #11: Chart not showing

no1dea, 2021-12-10:
First of all thank  your for this awesome work.

I updated today to the newest version first after having some issues with the temperature readings, now the chart on the Main Page is no longer showing. I think it's an issue with the log file location, at least that's what i was able to determine with my limited knowledge. I tried to grasp the interactions between the PIDKiln_vars.json, chart.js and the PIDKiln_logs.ino. 

The Chrome developer tools show an Error 500 when the charts.js tries to read the .csv, as it is looking for it in the current folder and not in the logs folder. 

`GET http://192.168.51.138/211210_105721.csv 500 (Internal Server Error)
request @ chartjs-datasource.min.js:7
beforeUpdate @ chartjs-datasource.min.js:7
notify @ Chart.2.9.3.bundle.min.js:7
update @ Chart.2.9.3.bundle.min.js:7
chart_update @ chart.js:163
setTimeout (async)
chart_update @ chart.js:164
Uncaught TypeError: Cannot read properties of undefined (reading 'datasets')
    at chartjs-datasource.min.js:7:9686
    at n.<anonymous> (chartjs-datasource.min.js:7:10087)
    at XMLHttpRequest.n.onreadystatechange (chartjs-datasource.min.js:7:775)`

If you need any more information I'd be happy to help.

Saur0o0n, 2021-12-10:
Hi,
What is the serial output debug? The chart will not show unless you load the program and there are data to be seen.

no1dea, 2021-12-10:
I have to rewire the controller so it's just powered by usb, but the kiln is running and also producing an .cvs file which I can open, the chart is just not showing the current running program and the mentioned error is in the chrome console log.

I'll get back to you with a debug output, hopefully tomorrow.

Saur0o0n, 2021-12-10:
What ESP32 version framework do you have right now? With PIDKiln 1.2 you have to have 2.0.x - it will not work with older ones (since there was SPIFFS directory handling change).
You can see it in Board Manager:
[image]

no1dea, 2021-12-10:
Thank your for your really fast answer! 

Yes I updated all the libraries and board managers to the newest versions beforehand and everything compiles fine. I attached a debug log, but i don't think there is anything in it. 
[pidkiln-debug.log](https://github.com/Saur0o0n/PIDKiln/files/7695774/pidkiln-debug.log)

But I found something strange, when I load the program, the chart is showing until the program is started, then it's gone.

djObsidian, 2022-01-27:
Same problem
[image]
Chart is not showing when programm is running

djObsidian, 2022-01-27:
chart.js also differs
https://github.com/Saur0o0n/PIDKiln/blob/master/data/js/chart.js#L97
But if i request it from esp
GET 192.168.2.78/js/chart.js
it looks different:
`plugins: {
			datasource: {
				type: 'csv',
				url: '/logs/test.csv',
				delimiter: ',',
				rowMapping: 'index',
				datasetLabels: true,
				indexLabels: true
			}
		}`
Isn't that weired?

Saur0o0n, 2022-01-27:
I have to sit on it, perhaps there are some other changes on ESP2.0.x framework, not only SPIFFS handling, that mess with file processing.

djObsidian, 2022-01-27:
I'm actually not even sure if this is backend problem. 
While programm is running,
`GET http://192.168.2.78/220127_112559.csv`
returns `HTTP/1.1 500 Internal Server Error  `
But, 
`GET http://192.168.2.78/logs/220127_112559.csv`
Returns  `HTTP/1.1 200 OK `

The file is there, but we are requesting wrong path for some reason. I didn't investigated how your code works, but here are some thougths:
1) Since `GET 192.168.2.78/js/chart.js`
has `url: '/logs/test.csv'`
it implies that CSVFile is null, while it actually shouldn't
https://github.com/Saur0o0n/PIDKiln/blob/master/PIDKiln_http.ino#L333
2) Adding propper error handling would be nice, because http error should be 404, not 500
3) Sidenote, https://github.com/Saur0o0n/PIDKiln/blob/master/PIDKiln_http.ino#L754
Versions of Chart.js are different for local file and loading it from internet. Not sure if that's significant, but with _Send large JS from local server_ setting set to _Redirect to Internet sources (recommended)_ chart does not works at all

djObsidian, 2022-01-27:
Okay, i found the issue:
https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#name
_Returns short (no-path) file name, as const char*. Convert it to String for storage._
So, all file.name should be replaced with file.fullName

Saur0o0n, 2022-01-27:
So as I said - it's  because ESP2.0.x changes in handling Spiffs - I must oversee it (see [this bug](https://github.com/Saur0o0n/PIDKiln/issues/10#issuecomment-972815240))

djObsidian, 2022-01-27:
https://github.com/espressif/arduino-esp32/releases/tag/2.0.0
https://github.com/espressif/arduino-esp32/commit/f6c9faf4daeff4ac6bcfafd55dbb1fb922b304ed
Should be as simple as changing .name to .path everywhere

Saur0o0n, 2022-01-27:
I'll take a look - it should be simple, but needs testing.

>     2. Adding propper error handling would be nice, because http error should be 404, not 500
This is done by webserver - probably there is something more going on than just filepath.

> 3. Sidenote, https://github.com/Saur0o0n/PIDKiln/blob/master/PIDKiln_http.ino#L754
>        Versions of Chart.js are different for local file and loading it from internet. Not sure if that's [...]

Only 3 files can be served externally
```
  if(Prefs[PRF_HTTP_JS_LOCAL].value.str){
    server.on("/js/jquery-3.4.1.js", HTTP_GET, [](AsyncWebServerRequest* request) {
      AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/js/jquery-3.4.1.js", "text/javascript");
      response->addHeader("Content-Encoding", "gzip");
      request->send(response);
    });
    server.on("/js/Chart.2.9.3.bundle.min.js", HTTP_GET, [](AsyncWebServerRequest* request) {
      AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/js/Chart.2.9.3.bundle.min.js", "text/javascript");
      response->addHeader("Content-Encoding", "gzip");
      request->send(response);
    });
    server.on("/js/chartjs-datasource.min.js", HTTP_GET, [](AsyncWebServerRequest* request) {
      AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/js/chartjs-datasource.min.js", "text/javascript");
      response->addHeader("Content-Encoding", "gzip");
      request->send(response);
    });
  }else{
    server.on("/js/jquery-3.4.1.js", HTTP_GET, [](AsyncWebServerRequest* request) {
      request->redirect(JS_JQUERY);
    });
    server.on("/js/Chart.2.8.0.bundle.min.js", HTTP_GET, [](AsyncWebServerRequest* request) {
      request->redirect(JS_CHART);
    });
    server.on("/js/chartjs-datasource.min.js", HTTP_GET, [](AsyncWebServerRequest* request) {
      request->redirect(JS_CHART_DS);
    });
  }
```

djObsidian, 2022-01-27:
Note, only PIDKiln_http.ino requires to change all instances of .name to .path, other files should be left as is.
https://github.com/Saur0o0n/PIDKiln/blob/37ab4be20f6a670d46bc488bf3013dfc6d93d8c6/PIDKiln_http.ino#L333

https://github.com/Saur0o0n/PIDKiln/blob/37ab4be20f6a670d46bc488bf3013dfc6d93d8c6/PIDKiln_http.ino#L438

https://github.com/Saur0o0n/PIDKiln/blob/37ab4be20f6a670d46bc488bf3013dfc6d93d8c6/PIDKiln_http.ino#L504

https://github.com/Saur0o0n/PIDKiln/blob/37ab4be20f6a670d46bc488bf3013dfc6d93d8c6/PIDKiln_http.ino#L645    

I changed that and everything is working so far.

> This is done by webserver - probably there is something more going on than just filepath.

No, if i request any file that doesn't exists, it returns 500, not 404. 
Even example for server has code for not found handling
https://github.com/me-no-dev/ESPAsyncWebServer/blob/1d46269cedf477661ca8a29518414f4b74e957d4/examples/simple_server/simple_server.ino#L25

> Only 3 files can be served externally

I mean here
https://github.com/Saur0o0n/PIDKiln/blob/37ab4be20f6a670d46bc488bf3013dfc6d93d8c6/PIDKiln_http.ino#L754
and here
https://github.com/Saur0o0n/PIDKiln/blob/37ab4be20f6a670d46bc488bf3013dfc6d93d8c6/PIDKiln_http.ino#L768
different versions of Chart.min.js are specified

no1dea, 2022-01-28:
I can confirm that changing `.name()` to `.path()`  in `PIDKiln_http.ino` fixes the issue with the chart not showing while the kiln is running. 

Thank you for investigating @djObsidian and @Saur0o0n .

Saur0o0n, 2022-01-28:
Yah, I know - l've checked it yesterday, but I wanted to test it on full device (that I don't have at home) - not only on esp32 board, like previous bugfixes and changes. I've pushed those changes (and some memory stuff) - but there is more...

Saur0o0n, 2022-01-28:
I'm closing this one - I've pushed checked code, with some other fixes.

closed 2022-01-28 by Saur0o0n