Saur0o0n/PIDKiln #13: MAX31855 Linearization

djObsidian, 2022-01-21:
According to [datasheet](https://datasheets.maximintegrated.com/en/ds/MAX31855.pdf), page 8, max31855 assumes a linear relationship between temperature and voltage, which is not accurate. For accurate results linearisztion needs to be performed. Otherwise there will be temperature error >10 degrees at real TC temperatures greater than 200 degrees.
Here is an example of arduino code for linearising max31855 data
https://github.com/heypete/MAX31855-Linearization

djObsidian, 2022-01-21:
This is a serious issue. I, not being aware of the need for additional data correction, even contacted Maxim Integrated tech support. I measured the readings of max31855 and compared them with the expected ones. As you can see from the table below, the error is significant. 

t ref | mV theor. | mV measured | t max31855
-- | -- | -- | --
26 | 0 | 0 | 27
100 | 2.09 | 2.0 | 84
200 | 5.23 | 5.5 | 177
300 | 8.66 | 8.6 | 261
350 | 10.45 | 10.4 | 326

The response from tech support was following:

> 
> While the MAX31855 does perform automatic cold-junction compensation, it does not perform any linearization calculations. Linearizing the thermocouple results needs to be performed externally using the NIST calculations and coefficients: http://srdata.nist.gov/its90/main/
> 
> As an alternative, the MAX31856 performs on board linearization for all supported thermocouple types.

Saur0o0n, 2022-01-21:
Well, to be honest I had no idea if this is significant. Previously I've used modified Adafruit MAX31856 library that had additional linearization (not by default - also mod)
```
double Adafruit_MAX31855::linearizeCelcius(double internalTemp, double rawTemp) {
  //////////////////////////////////////////////////////////////////////////
  // Function:       Takes the thermocouple temperature measurement and
  //                   corrects the linearity errors.
  // Input:
  //   double internalTemp  = value from readInternal()
  //   double rawTemp       = value from readCelsius()
  //
  // Returns:
  //   double               = Corrected Celsius thermocouple temperature
  //////////////////////////////////////////////////////////////////////////
  // Adapted from https://github.com/heypete/MAX31855-Linearization
  // Equations and coefficients from: https://srdata.nist.gov/its90/main/its90_main_page.html
  // Retrieved 2018-09-01
```
This this library was less reliable and trow different errors - also it introduced at some point, some additional dependant libraries (Adafruit) that I thought are too much and complicated everything beyond usefulness.

djObsidian, 2022-01-21:
As far as I can see the library you are using now doesn't has any linearization. IMHO, that's significant, because you don't want your kiln to read 950 degrees when it's actually 1050. At temperatures below 300 degrees, that might be not a big deal

Saur0o0n, 2022-01-24:
It should be relatively easy to add it to library - it's just math - but since author still haven't accept my last commit, I've asked if we can [add it](https://github.com/enjoyneering/MAX31855/issues/8).

ruffner, 2024-09-17:
Looks like it would be straight forward to add this snipped in, since enjoyneering hasn't addressed your PR yet. https://learn.adafruit.com/calibrating-sensors/maxim-31855-linearization.