Saur0o0n/PIDKiln #15: PID is not working as expected thus untunable

djObsidian, 2022-01-31:
The implementation of the PID controller in the library used is completely strange, if not inadequate. I have never seen such a variation of the regulator, and I have not seen any method of tuning it.
https://github.com/br3ttb/Arduino-PID-Library/blob/9b4ca0e5b6d7bab9c6ac023e249d6af2446d99bb/PID_v1.cpp#L83
What the hell is this? 
I read his blog, I am aware of the benefits described, including for PonM, but there are simply no methods of setting up such a regulator other than a stupid search for values. No conventional methods, such as the Ziegler-Nichols Method, just don't work here. 
The author doesn't provide any tuning method on the blog or on the github. 
I tried to synthesise PID values using Matlab PID tuner and they didn't worked, not even close. 
First method from this [video ](https://www.youtube.com/watch?v=9vZSw_xzC9E) produced values somewhat close to matlab's ones, neither they worked. 
It would be nice to add a disclaimer about this in the README or replace the library with another, that implements classic PID

Saur0o0n, 2022-01-31:
To be honest, I haven't give it much thoughts at theoretical level - just used and check if it works for me. If you have enough knowledge, can't you try to discuss it with the library author? Changing it (library) would require a lot of testing, much, much more then changing MAX library for example.

djObsidian, 2022-01-31:
That's a good idea, but this library is abandoned, its author isn't responding to issues for more than 4 years now

djObsidian, 2022-02-01:
Also, could you tell more about how did you tuned PID for your application?
[image]
None of these curves seems to be good.

Saur0o0n, 2022-02-01:
This are tries and errors on physical test stand (as described) - since those are low temp it was easy to to perform them on the bench. I haven't expect curve to match exactly the program, just close and without overheating. But on the other hand, I haven't give much reading on PID itself - just principles and what to expect. That's why I've used library - being kind of, end user for it and assuming it will work properly.
It was enough I've build my kilin and controller from scratch ;)
Do you have any particular, working PID literary in mind - that could replace this one? (just to take a look for now)

djObsidian, 2022-02-01:
That's a more complicated question than i thought initaly. There is 
https://github.com/GyverLibs/GyverPID
But it has two issues:
1) It is documented only in russian
2) All code is written in header files, which is not a problem, but seems wrong

I'll look for other libraries

djObsidian, 2022-02-01:
Side question. How often sould this message appear in serial console?
https://github.com/Saur0o0n/PIDKiln/blob/87db341e9dfde9f5992bd66f55080c5b18152272/PIDKiln_program.ino#L534
I thought, that it should appear every time, `PID.Compute();` is called. But with different PID time window set, it's appearing with same frequency. Is this a correct behaivour?

Saur0o0n, 2022-02-01:
This is just the debug message, program is recalculated every second (from interrupts) - but since this message was printer so often it clogged the output/debug screen, so I've made it to be printed every 10th second.

djObsidian, 2022-02-01:
So, this means, time window for PID can't be lower than 1000 ms?

Saur0o0n, 2022-02-01:
Why? This is just debug information - PID is recalculated every 10ms (because I've introduced recently delay(10) - previously it was without it - but there where sometimes issues and watchdog killing process)

The pid recalculation is below this debug message, and it's done in main for() loop
https://github.com/Saur0o0n/PIDKiln/blob/87db341e9dfde9f5992bd66f55080c5b18152272/PIDKiln_program.ino#L541

djObsidian, 2022-02-08:
Okay, i'm starting to understand, why i failed in calculating PID parametrs. 
Could you please clarify, how heater power is contolled? As i understand, there is something like a very slow PWM with frequency of 1 Hz or lower. But what is maximum value for PID output, and what is its resolution?

Saur0o0n, 2022-02-08:
In general yes, since we have only ON/OFF (no voltage regulation etc) and heating elements, of this power, does not exactly like to be on/off too fast (also electric grid safety does not like it, since there is often a spike) it should be done with decent window size. Since I've used SSR there is not big problem for it - but for mechanical relays - it would be also a death sentence.
Anyway, this "procedure" is straight from [example of the library](https://github.com/br3ttb/Arduino-PID-Library/blob/master/examples/PID_RelayOutput/PID_RelayOutput.ino).

djObsidian, 2022-02-12:
Well, finally i did it. I managed to setup PID for my kiln. I managed to get overshoot only around 4% (or 0.6% if we calculate in Kelvins). It took some simulation in Simulink, the realization that not all PID controller tuning approaches fit all systems, and suffering from the non-obviousness of the thing. 
I can write some sort of guide, how i achieved this, if you are interested.
In short, I completely missed the response time of the thermocouple, which plays a very big role in my case. I have a 3.6 liter, 1.8 kW oven, which can reach a staggering 5 degrees per second at temperatures up to 800-900 degrees. 

PID controller tuning methods can be divided into 2 categories: by transient response and by frequency response. And the first category, which I tried to use, does not work at all for systems with a delay in the output signal. Sure, kiln can be autotuned with this method https://github.com/br3ttb/Arduino-PID-AutoTune-Library
Or matlab closed loop PID tuner https://youtu.be/ENhFepqjFOY?t=156
It would be nice to have some form of autotuner in this project, but i'm not requesting it, it's requires a really good understanding of control theory and a hell lot of debugging, testing and simulation.

So, here's my model in Sumulink. Pretty straightforward, but i can explain it in details if you want me to write a guide. 

[image]

[image]

And this is test on real kiln
[image]
It requieres some fine tuning, but is already very close to ideal result, so it will take no more than a day instead of weeks of manual tuning.

Some notes:

1. I've had to replace this
https://github.com/Saur0o0n/PIDKiln/blob/87db341e9dfde9f5992bd66f55080c5b18152272/PIDKiln_addons.ino#L117
With this
`kiln_temp=kiln_tmp1;`
Because when heating up at full power difference between Last temp  and Average kiln temp was more than 10 degrees. I think, it would be better to measure temperature in a separate task with low period like 100 ms and use a proper Kalman filter. Also this will make easier to change temp sensor. I'm trying to use MAX31856, which is superior to 31855 (at the almost same price) but has conversion delay of 100 ms or more depending on settings.
2. [Example from PID library](https://github.com/br3ttb/Arduino-PID-Library/blob/master/examples/PID_RelayOutput/PID_RelayOutput.ino) is ok but there is really no point in calculating PID more often than once per WindowSize. So PID sample time can be set to WindowSize or WindowSize/2
3. My initial statement that PID is untunable because of the way it calculates derivative is completely wrong. In fact it hardly even changes behaivour of PID.