Saur0o0n/PIDKiln #16: Auto-tuning?

TurboNerd95, 2022-02-23:
I have 2 of those cheap crappy PID controllers like this one https://www.auberins.com/index.php?main_page=product_info&products_id=4. 

I've never bothered to try and tune my kilns manually, so I use their built in auto-tuning feature. It's not supposed to be very accurate, but it works find for my needs.

Have you considered maybe adding in the PID-auto-tuning library for Arduino? https://github.com/br3ttb/Arduino-PID-AutoTune-Library

I found one good example sketch of the auto-tuning library for a kiln here; https://www.megunolink.com/articles/malt-kiln/

Saur0o0n, 2022-02-28:
I've given some thought, during coding, about auto-tune. But, perhaps wrongly, I've assumed it's to big temperature scale (like 20C-1200C window) and too slow process to do reliable auto-tune. But I've assumed it without full understanding of autotune process.

TurboNerd95, 2022-08-02:
Hi, sorry for the late reply @pyr0ball .Adding autotune would be a really useful feature I think, so feel free to prototype it! From memory, the PIDautotune library requires about an hour to perform the Autotuning process. I think by following the example I linked, it would be quite easy to integrate it into this project, though I also don't have a great understanding of how PID works.

pyr0ball, 2022-08-06:
Thankfully @TurboNerd95, I've got a pretty decent amount of experience with heater PID tuning from my 3D printing. If I get stuck adding it to the project, would you mind if I reach out for assistance?