Saur0o0n/PIDKiln #18: Not aborting at Maximum housing temperature

stminm, 2022-05-31:
For testing purposes I set Maximum housing temperature to 30C in Safety features but PIDKiln is not aborting the program run when the temperature gets above the set value.
In logs I see when the temperature above 30 but the expected "Aborting program with error: #" is not following after:
... 1 - PIDKiln-ESP32 PIDKiln - - - ‹¯¨[ADDONS] Temperature sensor B readout: Internal temp = 30.2     Last temp = 33.8     Average case temp = 32.7

To notice, other settings in Safety features:  Minimal temperature and Maximum temperature are triggering abortion of the program run when going below or above set values respectively.

Saur0o0n, 2022-05-31:
It's checked once every ten seconds - so it does not have to appear after temperature readout.

stminm, 2022-05-31:
Yes, I have noticed how often it's checked but still, no program abort when exceeded.

Meanwhile I was doing some troubleshooting and here is how I fixed:
In `PIDKiln.h`
```
/* 
**  Program errors:
*/
typedef enum {
+ PR_ERR_HOUSE_HOT,       // exceeded max temperature defined in MAX_HOUS_TEMP
  PR_ERR_FILE_LOAD,       // failed to load file
...
```

In `PIDKiln_program.ino`
```
// Check all possible safety measures - if something is wrong - abort
//
void SAFETY_Check(){
  if(kiln_temp<Prefs[PRF_MIN_TEMP].value.uint8){
    DBG dbgLog(LOG_ERR,"[PRG] Safety check failed - MIN temperature < %d\n",Prefs[PRF_MIN_TEMP].value.uint8);
    ABORT_Program(PR_ERR_TOO_COLD);
  }else if(kiln_temp>Prefs[PRF_MAX_TEMP].value.uint16){
    DBG dbgLog(LOG_ERR,"[PRG] Safety check failed - MAX temperature > %d\n",Prefs[PRF_MAX_TEMP].value.uint16);
    ABORT_Program(PR_ERR_TOO_HOT);  
-  }
+  }else if(case_temp>Prefs[PRF_MAX_HOUS_TEMP].value.uint16){
+    DBG dbgLog(LOG_ERR,"[PRG] Safety check failed - MAX house temperature > %d\n",Prefs[PRF_MAX_HOUS_TEMP].value.uint16);
+    ABORT_Program(PR_ERR_HOUSE_HOT);
+  }
}
```

Saur0o0n, 2022-05-31:
Ahh, sorry I haven't notice you are talking about housing max temperature - I thought your are talking about kiln max temperature. So you are right - there was no checking for housing max temperature at all. What you propose as fix looks ok.

Saur0o0n, 2022-07-06:
Added to 1.4 - closing.

closed 2022-07-06 by Saur0o0n