Saur0o0n/PIDKiln #20: Alternative LCD's

pyr0ball, 2022-07-01:
Since the bare 12864B LCD's are becoming very difficult to find these days (due to the proliferation of RAMPS-style versions) would it be possible for you to include some support for the RAMPS-pinned version?

Another possible display I was considering was the SSD1309 OLED displays, if those could be made to work

Saur0o0n, 2022-07-06:
Hi,
It shouldn't be difficult to install RAMP 12864B (the one with sdcard and button) - it's all the same/similar hw, just perhaps controller is to be replaced in the code (but the display library will stay the same).
So fell free to experiment.
OLEDs display are to small to be useful (at least at my opinion) - and that's why I've chosen big (but low res) 128x64.

pyr0ball, 2022-07-12:
As I'm also using an alternative ESP carrier, would you be able to provide an example pinout for using the RAMPS LCD/SD/Encoder combo? That'll help immensely

pyr0ball, 2022-07-12:
found a reference!
[image]

I'll let you know how it goes!

pyr0ball, 2022-07-12:
OK I've nearly got it figured out, except for the CS pin for the SD Card, which isn't noted in your graphics since it's built into the module you've got

pyr0ball, 2022-07-16:
OK I think I figured out that the wiring for the unit you're using uses the SDcard in 4-bit mode, while that display's built-in sdcard is in SPI mode. I'm not super familiar with how to change that definition in your code. Any breadcrumbs you could provide?

Saur0o0n, 2022-07-16:
Hi,
I don't use SDCard in PIDKiln - only build in SPIFFS. And LCD is simple SPI interface - 4 wires only VSPI (MOSI-23, MISO-19, CLK-18, CS-5). Other LCD connections are for backlit and power.
After you figure out pins, you can change LCD type on the beginning of the PIDKiln_LCD.ino file
```
// You can switch hardware or software SPI interface to LCD. HW can be up to x10 faster - but requires special pins (and has some errors for me on 5V).
//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, /* clock=*/ LCD_CLOCK, /* data=*/ LCD_DATA, /* CS=*/ LCD_CS, /* reset=*/ LCD_RESET);
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R2, /* CS=*/ LCD_CS, /* reset=*/ LCD_RESET);
```

pyr0ball, 2022-07-18:
Gotcha, ok that just means that I'll have to add a different kind of sdcard reader with the 4 data lines and all that instead of using the reader baked into the LCD

pyr0ball, 2022-07-20:
derp I misunderstood that the SPIFFS filesystem was baked into the onboard SPI flash. I suppose SDCard could be used for some other features in future, but I'll just leave that for now.

I'll see if I can get this display working with SPI mode first, but I'll also wire up the data lines to allow 4-bit mode just in case