Saur0o0n/PIDKiln #24: Problems with temperatures above 1100 degC

JensBovbjergdk, 2022-07-25:
My Kiln controller works very well up to 1100 degC. But even with the Grace counter sat to 100, the program still aborts above 1100.
By looking at the Syslog, this is because the counter increases more than it decreases, and eventually reaches 100.
The temperature measurements continue to be valid though, even if there are more fails than good readings.
I therefore suggest that the Grace Counter is replaced by a Grace Timer. 
That is: The program only aborts if no valid readings are obtained in a Grace Period of x minutes.
Hope this makes sense and thanks for the good work.
Kind regards 
Jens

djObsidian, 2022-07-31:
This is probably an EMC problem. I did also encountered this. Surprisingly, there was no errors when heater was off. 
Quick fix is to change 
https://github.com/Saur0o0n/PIDKiln/blob/b9da27c5f9b2d874825d29ed0e71314416a40b97/PIDKiln_addons.ino#L119
to 
`if(TempA_errors>0) TempA_errors=0;`
I think the problem is that max31855 reading occurs on the same time when SSR is on, so that sould be avoided and measurement taken at the end of time window, not at start

Saur0o0n, 2022-07-31:
It's true that there is a lot of interference - so the good practice is to shield entire controller and ground it. This helped me in many other projects.
I still find the zeroing errors count a questionable practice (this may lead to quite wrong temp readouts and fail the whole program. Imagine that while having 1000C in kiln, the probe is having a lot of errors and the true read is possible one time per minute (1/60) and this readout is wrong - like suddenly it reads 900C or 1100C - what is possible - this ruins the whole task).