Saur0o0n/PIDKiln #26: Please consider porting to PlatformIO ( Can provide assistance)

pyr0ball, 2022-08-25:
Given there's a number of library dependencies that are required for this project to work, and it seems versioning is also relevant among those dependencies, I think it's a good idea to start working under a PlatformIO project where dependent libraries can have fully automated installation and management.

I'm happy to help with this if desired! I was about to create my own fork with extra features as well, but wanted to make sure I'm sticking to a platform you, as the original dev, are comfortable with.

Saur0o0n, 2022-08-25:
Fell free to commit platformio.ini file - I was working Code it with 3d printer Marlin fw, but somehow had not need to switch. Perhaps now is the chance ;)

pyr0ball, 2022-08-25:
Okipokey! I'll carve out some time to do this next week.

I just finished assembling my custom controller PCB, so we'll see how it goes ;) 
[image]

Saur0o0n, 2022-08-25:
I see MOSFETts - so this is for some low (like lower then kilowatts) power solution?

pyr0ball, 2022-08-25:
those two fets are for fan control. I plan on having one cooling the SSR's heatsink, and another for maybe a recirculation fan (as this controller could also be adapted for something like a reflow oven)

I've also added an extra SSR and EMR output for kilns like one of mine that has dual independent coils. All of those outputs are opto-couple isolated.

There's also a header for running a Neopixel (WS281x) LED strip (mainly just for bling, but this could be used for status indicators or all kinds of other things)

The two 10-pin headers are basically just mapped for 3D Printer controller LCD's so any displays that work with that style of connection can theoretically be used. I've chosen to go with the RepRapDiscount Full Graphic Smart controller as they're prolific and cheap and use the same display as the one you're originally using.

Next rev adds 5x GPIO pins for other stuff like a door switch (mentioned by another user in a different issue) or could be used for external physical buttons, a case thermistor input, or even get tied in with a smoke/gas detector for automatic shutoff. I'm also planning on migrating the EMR gate pin to somewhere else so we can take advantage of the I2C bus and maybe get a little extra feature-ability like maybe some current detection IC's. Another thing I would love to add would be support for the SSD1309 displays as a secondary informational output showing things like the the actual heater coil resistance, current power metrics like voltage and wattage etc so that all can be tracked. I believe with that data I could add in a maintenance detection feature or a "remaining health" meter for the coils so the user knows when it's time to get them replaced.

pyr0ball, 2022-08-26:
@Saur0o0n would it be possible for you to open up a `develop` branch for me to do my pull requests against?

Saur0o0n, 2022-08-27:
Hi,
I've created platformio branch (https://github.com/Saur0o0n/PIDKiln/tree/platformio)

djObsidian, 2022-08-27:
Wow, that's a great one PCB. I also designed my own, as a replacement for Rex C-100.
[image]

Porting this project to PlatformiO is a good idea. But refactoring and splitting code into several files as for arduino IDE would take some time. I ported for myself by simply throwing all code into main.c and declaring all functions in main.h. Not really convinient for writing and editing code but compiles like 10 times faster than in Arduino IDE. 

My platformio.ini looks like this. ESP32 in PIO is not still updated to Arduino Core 2.0 so it was a bit tricky to configure. 

```[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
lib_deps = 
	adafruit/Adafruit MAX31856 library@^1.2.5
	adafruit/Adafruit BusIO@^1.11.1
	Wire
	SPI
	br3ttb/PID @ ^1.2.1
	olikraus/U8g2@^2.33.2
	arcao/Syslog @ ^2.0.0
monitor_speed = 115200

pyr0ball, 2022-08-28:
Well that's a heck of a lot of work you've already done! If you are up to doing a PR on that new branch with your current refactor, I can work on the file segregation while I'm adding the other features

djObsidian, 2022-08-28:
I made a PR, you can check it out

HeintzelmannDK, 2024-11-02:
Did any of you ever make a PlatformIO version?

Saur0o0n, 2024-11-02:
djObsidian made a PR, but never finished it. I kind of never fully switched to PIO, so had not big interest in it.

HeintzelmannDK, 2024-11-02:
> platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git

Ok. Thanks for the quick reply.