Saur0o0n/PIDKiln #31: "File not found" when connected

masterjuggler, 2022-10-05:
I just went through the installation process on my esp32-wroom-32e. Everything compiled and uploaded without issue on the latest arduino ide. When I connect to the resulting "PIDKiln_AP" and enter the default ip 192.168.10.1 into a browser all I get is a blank page that says "file not found"

Am I just not connecting to it correctly? I didn't see anything on the wiki that mentions another way.

Saur0o0n, 2022-10-05:
Seems like did not upload the data to your Esp SPIFFS or you put something strange in URL after the IP address.

masterjuggler, 2022-10-05:
Alright, I finally managed to figure it out. It doesn't seem like arduino ide 2.0.0 has no support for plugins like esp32fs, so I had to grab 1.8.19 and that worked.

closed 2022-12-30 by Saur0o0n