Saur0o0n/PIDKiln #32: Temperature readings are far out and unstable

atlantis777, 2022-12-31:
Hi Adrian

I have contacted you about this by email and as you suggested I now open an issue here.

To show you what I meant with the readings being unstable I took a short video from the main view on the lcd. This unstableness does also show in the web view. The case temperature was stable this time but the inside and ambient temps were not. I'm not sure if the case temp was also unstable last time or not. The kiln was cold at the time so the values of all readings are far out. 

Furthermore I attached two logs, one of a program of yours and one of a program of mine. Please tell me what further info I can provide. 

Kind regards
Martin

[221231_123252.log](https://github.com/Saur0o0n/PIDKiln/files/10327264/221231_123252.log)
[221231_123745.log](https://github.com/Saur0o0n/PIDKiln/files/10327265/221231_123745.log)

https://user-images.githubusercontent.com/51721346/210135747-9ab9888d-b483-4ad7-8e24-23a9041a4ea9.mov

Saur0o0n, 2023-01-02:
So there is no way, this temperature changes could happen so fast on normal circumstances. The fact, it's not even showing proper temperature - makes me wander, if you simply don't have wrong/faulty max chip and/or connection issue (or perhaps no K thermocouple). In this case, I would load simple sketch for MAX6675 (from examples - like MAX6675_hw_SPI_Demo) and make it work first.

And btw. by "send some logs" I meant serial output of Pidkiln - not running program log :). In those serial logs (or syslog if you forward them) you will see actual readouts and values.

atlantis777, 2023-01-03:
Well of course it could be possible that both MAX6675 are damaged but it's rather unlikely I guess, unfortunately I don't have spares at the moment but have to order some first. I think I made some kind of mistake but I don't seem to find it, I re-checked the wiring etc. and it looks ok but there's always the possibility that I don't see the forest for the trees. Please describe how I take those serial logs if they're helpful, sorry that I misunderstood.

Saur0o0n, 2023-01-03:
So if MAX6675 would be wrongly connected, it should not be responding at all - this is digital communication. So more likely is something else. Disable all other thermocouples and leave only main, kiln one (disable it in Pidkiln and disconnect electrically). Run some test program and check if temperature is stable, if not - change thermocouple (physically - MAX6675 leave the same) and re try. To eliminate thermocouple issue, MAX6675 interference issues on SPI line etc.
And like I said - you should make thermocouple/MAX6675 work first without Pidkiln, it's not Pidkiln issue here.

atlantis777, 2023-01-09:
Didn't I say I must have overlooked something and so it was. The thermocouple boards I had installed had the MAX6675 on instead of the MAX31855. I was lucky enough to find a 31855 board in another device that I could borrow for a test and with that the kiln heated up and the readings made sense. So I ordered a few boards with the right IC on...

However I had two other issues when testing:

- Sometimes when navigating the menu the lcd went blank. In the background it looked like the controller was still working but the gui didn't come back. I guess this is a problem with either the lcd itself or the lcd comms.
- I had problems transferring program files with the names of "Nanovia_old.txt" and "Nanovia_new.txt". Both were rejected for bad filenames. "Nan_old.txt" and "Nan_new.txt" worked in the end and "Linkcast_small.txt", "Linkcast_medium.txt" and "Linkcast_large.txt" had all not been rejected either.

Saur0o0n, 2023-01-09:
On filenames - limit is 32 characters, directory name included (it's a SPIFFS limit). I'm not sure if this is the case - but serial debug output should give you the hint where the problem may be - if there is any.