Saur0o0n/PIDKiln #33: after sketch upload, program starts and constantly restarting

ggioorgi, 2023-03-05:
board: esp32-wroom-32D

serial

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
E (218) psram: PSRAM ID read error: 0xffffffff
[PREFS] Loading prefs from file[PREFS]  -> For WiFi_Retry_cnt saved UINT8 item value:9 type:1
[PREFS]  -> For WiFi_Mode saved UINT8 item value:1 type:1
[PREFS]  -> For HTTP_Local_JS saved UINT8 item value:1 type:1
[PREFS]  -> For Auth_Username saved STRING item value:admin type:4
[PREFS]  -> For Auth_Password saved STRING item value:hotashell type:4
[PREFS]  -> For NTP_Server1 saved STRING item value:0.pl.pool.ntp.org type:4
[PREFS]  -> For NTP_Server2 saved STRING item value:1.pl.pool.ntp.org type:4
[PREFS]  -> For NTP_Server3 saved STRING item value:2.pl.pool.ntp.org type:4
[PREFS]  -> For GMT_Offset_sec saved INT16 item value:3600 type:3
[PREFS]  -> For Daylight_Offset_sec saved INT16 item value:3600 type:3
[PREFS]  -> For Initial_Date saved STRING item value:2021-12-02 type:4
[PREFS]  -> For Initial_Time saved STRING item value:11:00:00 type:4
[PREFS]  -> For PID_Window saved UINT16 item value:5000 type:2
[PREFS]  -> For PID_Kp saved VFLOAT item value:20.000000 type:5
[PREFS]  -> For PID_Ki saved VFLOAT item value:0.200000 type:5
[PREFS]  -> For PID_Kd saved VFLOAT item value:0.100000 type:5
[PREFS]  -> For PID_Temp_Threshold saved INT16 item value:-1 type:3
[PREFS]  -> For LOG_Window saved UINT16 item value:30 type:2
[PREFS]  -> For LOG_Files_Limit saved UINT16 item value:40 type:2
[PREFS]  -> For DBG_Serial saved UINT8 item value:1 type:1
[PREFS]  -> For DBG_Syslog saved UINT8 item value:0 type:1
[PREFS]  -> For DBG_Syslog_Srv saved STRING item value:192.168.1.2 type:4
[PREFS]  -> For DBG_Syslog_Port saved UINT16 item value:514 type:2
[PREFS]  -> For MIN_Temperature saved UINT8 item value:10 type:1
[PREFS]  -> For MAX_Temperature saved UINT16 item value:1350 type:2
[PREFS]  -> For MAX_Housing_Temperature saved UINT16 item value:130 type:2
[PREFS]  -> For Thermal_Runaway saved UINT16 item value:0 type:2
[PREFS]  -> For Alarm_Timeout saved UINT16 item value:5 type:2
[PREFS]  -> For MAX31855_Error_Grace_Count saved UINT8 item value:5 type:1
[PREFS] -=-=-= PREFS DISPLAY =-=-=-[PREFS] 1) 'WiFi_SSID' = ''	4
[PREFS] 2) 'WiFi_Password' = ''	4
[PREFS] 3) 'WiFi_Mode' = '1'	1
[PREFS] 4) 'WiFi_Retry_cnt' = '9'	1
[PREFS] 8) 'HTTP_Local_JS' = '1'	1
[PREFS] 9) 'Auth_Username' = 'admin'	4
[PREFS] 10) 'Auth_Password' = 'hotashell'	4
[PREFS] 11) 'NTP_Server1' = '0.pl.pool.ntp.org'	4
[PREFS] 12) 'NTP_Server2' = '1.pl.pool.ntp.org'	4
[PREFS] 13) 'NTP_Server3' = '2.pl.pool.ntp.org'	4
[PREFS] 14) 'GMT_Offset_sec' = '3600'	3
[PREFS] 15) 'Daylight_Offset_sec' = '3600'	3
[PREFS] 16) 'Initial_Date' = '2021-12-02'	4
[PREFS] 17) 'Initial_Time' = '11:00:00'	4
[PREFS] 18) 'PID_Window' = '5000'	2
[PREFS] 22) 'PID_POE' = '0'	1
[PREFS] 23) 'PID_Temp_Threshold' = '-1'	3
[PREFS] 24) 'LOG_Window' = '30'	2
[PREFS] 25) 'LOG_Files_Limit' = '40'	2
[PREFS] 26) 'MIN_Temperature' = '10'	1
[PREFS] 27) 'MAX_Temperature' = '1350'	2
[PREFS] 28) 'MAX_Housing_Temperature' = '130'	2
[PREFS] 29) 'Thermal_Runaway' = '0'	2
[PREFS] 30) 'Alarm_Timeout' = '5'	2
[PREFS] 31) 'MAX31855_Error_Grace_Count' = '5'	1
[PREFS] 32) 'DBG_Serial' = '1'	1
[PREFS] 33) 'DBG_Syslog' = '0'	1
[PREFS] 34) 'DBG_Syslog_Srv' = '192.168.1.2'	4
[PREFS] 35) 'DBG_Syslog_Port' = '514'	2
WiFi mode: 1, Retry count: 9, is wifi enabled: 4
[NET] Y:2021 [NET] M:12 [NET] D:02 [NET] H:11 [NET] m:00 [NET] s:00
[NET] Creating WiFi Access Point (AP)192.168.10.1
[PRG] Loading directory...
[PRG]	counted 6 files
[PRG] Processing filename: program1.txt
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008b9fa  PS      : 0x00060830  A0      : 0x800daa91  A1      : 0x3ffb2620  
A2      : 0x00000000  A3      : 0x3ffb2690  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x676f7270  A9      : 0x00000069  
A10     : 0x00000000  A11     : 0x3ffd55b8  A12     : 0x3f401dd2  A13     : 0x3ffb2610  
A14     : 0x3ffb25e0  A15     : 0x00000008  SAR     : 0x00000012  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008b9ed  LEND    : 0x4008ba05  LCOUNT  : 0xffffffff  

Backtrace:0x4008b9f7:0x3ffb26200x400daa8e:0x3ffb2630 0x400dac58:0x3ffb26d0 0x400db484:0x3ffb27b0 0x400ecea2:0x3ffb2820 

ELF file SHA256: 0000000000000000

Saur0o0n, 2023-03-05:
What esp framework version do you have?

ggioorgi, 2023-03-06:
1.0.6 version

I delete all files in 'programs' folder and restarting stop after fresh install. I open web interface, it reads temps, but can't write any data, when upload one program, same thing happened and start restarting loop with same error

ggioorgi, 2023-03-06:
[image]

ggioorgi, 2023-03-06:
[image]

Saur0o0n, 2023-03-06:
Did you change the settings for Wroom (default is Wrover) - the wroom chips doesn't have psram.
https://github.com/Saur0o0n/pidkiln/wiki/PIDKiln-hardware-components

ggioorgi, 2023-03-06:
yes I change that

ggioorgi, 2023-03-06:
I tried to change many things, nothing helps. Maybe my board is damaged, i'll try another board. I will buy exactly the board you have.

ggioorgi, 2023-03-06:
Ok now I installed another program that also writes and uses the files and it works fine. Your program gives me an error when it needs to write or read from memory. But is better buy new board which I know will work, instead of change program.

Saur0o0n, 2023-03-06:
I think it's the esp framework - that's why I've asked for it's version. Please try a bit older one like 1.0.4 - they have changed a lot at some point with file handling on SPIFFS (https://github.com/Saur0o0n/PIDKiln/issues/10#issuecomment-972815240). I'll have to compile it later with newest one and check it for myself.

ggioorgi, 2023-03-06:
I try 1.0.3 and 1.0.4 versions. same error.

Saur0o0n, 2023-03-07:
Hmm,
At this point - I don't have idea of the common problem here. Unless you did something wrong with spiffs formatting.
You can decode your Guru Meditation error - but it has to be on your binaries/compilation (I can't do that):
[image]
And send the output back.

closed 2024-02-09 by Saur0o0n