Saur0o0n/PIDKiln #36: Are there still issues with filesystem? Chart not showing

astern1, 2023-04-21:
Hi,

I'm just searching for a kiln control and came along, looks very great, thanks a lot for that!
Unfortunately I can't get it running properly.
I tried with WROM and Wrover Board but with both I get a similar failure like https://github.com/Saur0o0n/PIDKiln/issues/11
Config for the PSRAM was done properly, I can also upload own txt files as profiles, they show up in the list (via webfrontend as well as via Sketch uploader in Arduino IDE)

So main view is empty, when I click on a profile to load it it's like nothing happens.
[image]

Also the Debug output looks very weird:
[image]

Same for preferences:
[image]

Arduino IDE I'm using is 1.8.19
esp32 I tried a lot, 2.0.8 | 2.0.1 | 1.0.4 and so on...
[image]

Any suggestions what to try?

Saur0o0n, 2023-04-21:
Hi,
Check https://github.com/Saur0o0n/pidkiln/wiki/Required-software - look for ~ :)

astern1, 2023-05-18:
Seems like it ewas an error during saving of the WebResponseImpl.h

Thanks!