Saur0o0n/PIDKiln #40: Q; Control 2 heaters using just one PIDKiln?

ilgrank, 2023-06-15:
Hi
This is  question (not an issue): is it possible to use PIDKiln to control an oven with two (one upper, one lower) indipendent heaters?
Thanks!

Saur0o0n, 2023-06-15:
Yes,  you can - it's still enabled at the same time, but it can have different/independant power source.
```/*
** Relays and thermocouple defs. and other addons
**
*/

#define EMR_RELAY_PIN 21
#define SSR1_RELAY_PIN 19
//#define SSR2_RELAY_PIN 22   // if you want to use additional SSR for second heater, uncoment this
```

ilgrank, 2023-06-21:
and that's true even for a second temperature sensor?

Saur0o0n, 2023-06-24:
Hi, nope - as I said, they act as a single heater but with two separated power sources. And there is only single temperature sensor for kiln interior - other sensor is for outside temperature.

closed 2024-02-09 by Saur0o0n