Saur0o0n/PIDKiln #41: Cannot compile: redefinition of 'class IPv6Address

videobuff, 2023-07-08:
Cannot compile

System: ESP-WROOM-32
Arduino on line IDE

Reinstalled new development with Arduino online coding.
Imported the sketch.
I get errors complaining "redefinition of 'class IPv6Address"

```
In file included from /home/builder/Arduino/libraries/ethersia_2_4_0/src/IPv6Packet.h:11,
from /home/builder/Arduino/libraries/ethersia_2_4_0/src/UDPSocket.h:10,
from /home/builder/Arduino/libraries/ethersia_2_4_0/src/Syslog.h:9,
from /tmp/437171178/PIDKiln/PIDKiln.h:2,
from /tmp/437171178/PIDKiln/PIDKiln.ino:35:
/home/builder/Arduino/libraries/ethersia_2_4_0/src/IPv6Address.h:27:7: error: redefinition of 'class IPv6Address'
class IPv6Address {
```

Saur0o0n, 2023-07-10:
It looks like you should uninstall that ethersia library.

videobuff, 2023-07-10:
I will try to do so but so far it is not listed in the IDE. I  went back to version 1.8.9 I believe an that one compilesVerstuurd vanaf mijn telefoon

closed 2023-07-17 by Saur0o0n