Saur0o0n/PIDKiln #44: MAX6675  instead of MAX31855

ggioorgii, 2023-11-21:
first of all thank you for this great project.
In my local area I only find  MAX6675 boards. It is possible to easily replace code and use MAX6675 somehow?
Or order MAX31855 and wait 1 month ? :(

Saur0o0n, 2023-11-21:
I remember I had a brief discussion with somebody on 6675 (https://github.com/Saur0o0n/PIDKiln/issues/32#issue-1515106087) - but don't remember if I've checked then if this is feasible to use it. Also quick look shows it's 0°C to +1024°C - so you also have to take it into consideration.

ggioorgii, 2023-11-22:
I find Max31855 and setup everything. It almost works. max6675 give me 1000c reads always.

But I have new problem. I can on/off relay by start or stop program, but when I set temp by program, it not try to hold that temp. relay  is always on and kiln temp is increasing nonstop.

ggioorgii, 2023-11-22:
[image]

ggioorgii, 2023-11-23:
pin 19 was problematic. I don't know why. I using one relay, so uncomment second relay, set first relay to pin22 now everything works.

ggioorgii, 2023-11-23:
I am so happy with this. Thank you again. little bit hard to setup for me, because I using different hardware, but it worth it.