Saur0o0n/PIDKiln #48: Compilation error: invalid conversion from 'const AsyncWebHeader*' to 'AsyncWebHeader*' [-fpermissive]

kingnoob1377, 2024-11-17:
I had errors compiling pidkiln.
I had other several errors compiling, but I was able to get through them somehow.
This thing I have no idea.

how can I complete my compilation?

compilation terminal output
[log.txt](https://github.com/user-attachments/files/17790503/log.txt)

installed libraries
[image]
 
My sketch
[PIDKiln.zip](https://github.com/user-attachments/files/17790519/PIDKiln.zip)

Saur0o0n, 2024-11-17:
Hi, I would guess there are some braking changes both in esp32 Arduino framework and asyncwebserver. You can try to compile it with older versions. I'll also probably update/fix the code, but can't say now when exactly :)

kingnoob1377, 2024-11-24:
thanks for the reply. downgrading esp32 board library fixed it, but I also had to downgrade the arduino ide due to the ping timout error. now I can upload the sketch without an issue.

Saur0o0n, 2024-12-18:
I've updated the code to compile with recent esp and libraries - although it's not tested much.

closed 2024-12-18 by Saur0o0n