Saur0o0n/PIDKiln #49: Runtime issue

terrin2k, 2024-12-15:
Hey everyone!

So I can build and upload the code & filesystem to my ESP32 and it runs... however, I get the following endless panic/reboot cycle:

```
<snip>
[PREFS] 31) 'MAX31855_Error_Grace_Count' = '5'	1
[PREFS] 32) 'DBG_Serial' = '1'	1
[PREFS] 33) 'DBG_Syslog' = '0'	1
[PREFS] 34) 'DBG_Syslog_Srv' = '192.168.1.2'	4
[PREFS] 35) 'DBG_Syslog_Port' = '514'	2
WiFi mode: 1, Retry count: 9, is wifi enabled: 4
[NET] Y:2024 [NET] M:01 [NET] D:01 [NET] H:11 [NET] m:00 [NET] s:00
[NET] Connecting to WiFi as Client...
[LOG] Syslog disabled
192.168.0.100
[PRG] Loading directory...
[PRG]	counted 6 files
[PRG] Processing filename: program1.txt
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c15a  PS      : 0x00060730  A0      : 0x800daba1  A1      : 0x3ffb2080  
A2      : 0x00000000  A3      : 0x3ffb20f0  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x676f7270  A9      : 0x00000069  
A10     : 0x00000000  A11     : 0x3ffd94b8  A12     : 0x3f401dba  A13     : 0x3ffb2070  
A14     : 0x3ffb2040  A15     : 0x00000008  SAR     : 0x00000016  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008c14d  LEND    : 0x4008c161  LCOUNT  : 0xffffffff  

Backtrace: 0x4008c157:0x3ffb2080 0x400dab9e:0x3ffb2090 0x400dadb3:0x3ffb2130 0x400db593:0x3ffb2210 0x400f096f:0x3ffb2290
```

Decoded backtrace shows this:
```
    0x4008c157: strcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcpy.S:103
    0x400dab9e: Load_programs_dir() at C:\Users\danie\Documents\Arduino\PIDKiln\PIDKiln_program.ino:157
    0x400dadb3: Generate_INDEX() at C:\Users\danie\Documents\Arduino\PIDKiln\PIDKiln_http.ino:204
    0x400db593: setup() at C:\Users\danie\Documents\Arduino\PIDKiln\PIDKiln.ino:169
    0x400f096f: loopTask(void*) at C:\Users\danie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32\main.cpp:58
```

I am using:
- Arduino IDE 1.8.18
- ESP32 board package 3.0.1 (also tried latest 3.0.7, and various 2.0.X which fail with a build error)
- Libraries:
- AsyncTCP 3.2.15
- ESP Async Webserver 3.4.0
- PID 1.2.0
- SimpleSyslog 0.1.2
- U8G2 2.35.30

I've tested this with my ESP32 board (ESP-WROVER-32U based) isolated from the hardware and in-situ and it crashes either way.

I've verified that I have the 4MB board and the correct Program/SPIFFs profile chosen in the Arduino IDE.

Any ideas as to what might be going on here?

Thanks!

~ D

Saur0o0n, 2024-12-15:
Hi, Since it's failing on strcpy it looks like some memory allocation issue. Are you sure you have wrover (not wroom) board? You can try to change #define settings for wroom memory allocation and try again.
I'll try to find some time during holidays to update the code to make it work with current arduino/esp32 IDE.

terrin2k, 2024-12-15:
Argh. You are 100% correct. I have an ESP32-WROOM-32U. <facepalm>

I'm rebuilding now. I'll let y'all know how it goes.

Thanks for the quick reply!

~ D

terrin2k, 2024-12-15:
Hmm. 

The ESP32 board I have has an ESP32-WROOM-32U on it, and 'ESP32_DevKitC_V4' on the silkscreen on the back.

I've tried a couple of different boards from the IDE and I've completely erased the flash for each before trying to run the code. Each build still crashes as I showed above.

I've also tried another board of the same type with the same results.

Any other ideas?

~ D

```
[PREFS] 25) 'LOG_Files_Limit' = '40'	2
[PREFS] 26) 'MIN_Temperature' = '10'	1
[PREFS] 27) 'MAX_Temperature' = '1350'	2
[PREFS] 28) 'MAX_Housing_Temperature' = '130'	2
[PREFS] 29) 'Thermal_Runaway' = '0'	2
[PREFS] 30) 'Alarm_Timeout' = '5'	2
[PREFS] 31) 'MAX31855_Error_Grace_Count' = '5'	1
[PREFS] 32) 'DBG_Serial' = '1'	1
[PREFS] 33) 'DBG_Syslog' = '0'	1
[PREFS] 34) 'DBG_Syslog_Srv' = '192.168.1.2'	4
[PREFS] 35) 'DBG_Syslog_Port' = '514'	2
WiFi mode: 1, Retry count: 9, is wifi enabled: 4
[NET] Y:2024 [NET] M:01 [NET] D:01 [NET] H:11 [NET] m:00 [NET] s:00
[NET] Connecting to WiFi as Client...
[LOG] Syslog disabled
192.168.0.100
[PRG] Loading directory...
[PRG]	counted 6 files
[PRG] Processing filename: program1.txt
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008b32e  PS      : 0x00060f30  A0      : 0x800da99d  A1      : 0x3ffb2060  
A2      : 0x00000000  A3      : 0x3ffb20d0  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x676f7270  A9      : 0x00000069  
A10     : 0x00000000  A11     : 0x3ffd8764  A12     : 0x3f401dba  A13     : 0x3ffb2050  
A14     : 0x3ffb2020  A15     : 0x00000008  SAR     : 0x00000012  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008b321  LEND    : 0x4008b335  LCOUNT  : 0xffffffff  

Backtrace: 0x4008b32b:0x3ffb2060 0x400da99a:0x3ffb2070 0x400daba3:0x3ffb2110 0x400db37f:0x3ffb21f0 0x400efd77:0x3ffb2270 0x40091462:0x3ffb2290
```

terrin2k, 2024-12-20:
A little progress to report...

So I replaced my ESP32 boards with ones ordered from Espressif. These are DevKitC boards with an actual ESP32-WROVER-IE module onboard. When I run the code, I no longer get the strcpy crash. Yay!

Instead, the code runs for a while then crashes due to the WDT. Boo.
```
[PRG] FName: test_up_down.txt	 FSize:270	Sel:0
[PRG] Processing filename: vfail1.txt
[PRG] FName: vfail1.txt	 FSize:246	Sel:0
[PRG] Processing filename: vfail2.txt
[PRG] FName: vfail2.txt	 FSize:233	Sel:0
[PRG] Processing filename: index.html
[HTTP] Head of index - copying...
<tr><td><img src="/icons/heat.png" alt="[ICO]"></td><td><a href="/programs/program1.txt" target="_blank">program1.txt</a></td><td>491</td><td># Short, clear description of the program.</td><td><a href=/load.html?prog_name=program1.txt>load</a></td><td><a href=/delete.html?prog_name=program1.txt>delete</a></td></tr>
<tr><td><img src="/icons/heat.png" alt="[ICO]"></td><td><a href="/programs/program2.txt" target="_blank">program2.txt</a></td><td>143</td><td># Second program short description</td><td><a href=/load.html?prog_name=program2.txt>load</a></td><td><a href=/delete.html?prog_name=program2.txt>delete</a></td></tr>
<tr><td><img src="/icons/heat.png" alt="[ICO]"></td><td><a href="/programs/test_lowtemp.txt" target="_blank">test_lowtemp.txt</a></td><td>304</td><td># Program to test low temperatures</td><td><a href=/load.html?prog_name=test_lowtemp.txt>load</a></td><td><a href=/delete.html?prog_name=test_lowtemp.txt>delete</a></td></tr>
<tr><td><img src="/icons/heat.png" alt="[ICO]"></td><td><a href="/programs/test_up_down.txt" target="_blank">test_up_down.txt</a></td><td>270</td><td># Program to test up and down temp. change</td><td><a href=/load.html?prog_name=test_up_down.txt>load</a></td><td><a href=/delete.html?prog_name=test_up_down.txt>delete</a></td></tr>
<tr><td><img src="/icons/heat.png" alt="[ICO]"></td><td><a href="/programs/vfail1.txt" target="_blank">vfail1.txt</a></td><td>246</td><td># Short description of fail1</td><td><a href=/load.html?prog_name=vfail1.txt>load</a></td><td><a href=/delete.html?prog_name=vfail1.txt>delete</a></td></tr>
<tr><td><img src="/icons/heat.png" alt="[ICO]"></td><td><a href="/programs/vfail2.txt" target="_blank">vfail2.txt</a></td><td>233</td><td># Short description of fail2</td><td><a href=/load.html?prog_name=vfail2.txt>load</a></td><td><a href=/delete.html?prog_name=vfail2.txt>delete</a></td></tr>
[HTTP] End of index - copying...
[LOG] Loading dir: Loading logs directory...
[LOG] Loading dir:	counted 1 files
[LOG] Loading dir: Loading logs directory...
[LOG] Loading dir:	counted 1 files
[HTTP] Head of logs - copying...
[HTTP] End of log index - copying...
[ADDONS] Temperature sensor A readout: Internal temp = 20.7 	 Last temp = 22.8 	 Average kiln temp = 20.3
[ADDONS] Temperature sensor A readout: Internal temp = 21.0 	 Last temp = 22.8 	 Average kiln temp = 20.5
[ADDONS] Temperature sensor A readout: Internal temp = 21.1 	 Last temp = 22.8 	 Average kiln temp = 20.7
[ADDONS] Temperature sensor B readout: Internal temp = 21.2 	 Last temp = 22.5 	 Average case temp = 20.5
[ADDONS] Temperature sensor A readout: Internal temp = 21.2 	 Last temp = 22.8 	 Average kiln temp = 20.9
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.1
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.3
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.4
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.6
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.7
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.8
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 21.9
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.0
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.1
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.1
[ADDONS] Temperature sensor B readout: Internal temp = 21.2 	 Last temp = 22.5 	 Average case temp = 20.9
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.2
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.2
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.3
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.3
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.4
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.4
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.4
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.5
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.5
[ADDONS] Temperature sensor A readout: Internal temp = 21.3 	 Last temp = 22.8 	 Average kiln temp = 22.5
[... cut ...]
E (18739) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (18739) task_wdt:  - IDLE0 (CPU 0)
E (18739) task_wdt: Tasks currently running:
E (18739) task_wdt: CPU 0: Input_loop
E (18739) task_wdt: CPU 1: loopTask
E (18739) task_wdt: Aborting.
E (18739) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4009110c:0x3ffb4d30 0x400ed3f3:0x3ffb4d50 0x400dc000:0x3ffb4d70 0x40090c52:0x3ffb4d90
```

I'll try disabling the WDT, but that's not a great permanent solution.

Thanks!

~ D

terrin2k, 2024-12-21:
Success!

In PIDkiln_input.ino, in Input_Loop(), I had to replace the call to yield() with vTaskDelay(1)... and now it all works!

Tomorrow afternoon, I'm going to start wiring this thing to the kiln and give it a go. :)

Thanks for everything!

~ D

terrin2k, 2024-12-21:
I'm marking this as closed.

closed 2024-12-21 by terrin2k