Saur0o0n/PIDKiln #50: Rebooting when trying to connect to WiFi

jvswan, 2025-01-05:
First, THANKS for this project! I'm setting it up for my daughter who acquired an old kiln with a non-working kiln sitter. We're excited to get it up and going!

The issue I'm having is that after installing PIDKiln on my ESP 32 WROOM, the system starts with the v1.5 splash screen and a brief message that says "starting..." After about a second it says "connecting WiFi..." Then, a second later the screen flashes and it goes back to "starting..." followed by "connecting WiFi..." I deleted my SSID and password from the pidkiln.conf file and changed the WiFi_Mode to 3, but it still goes through the same sequence. 

Differences in hardware include using the ESP32 WROOM board, a project breakout board, and I use two SSRs and two EMRs (I'm controlling 240v, so I have a relay for each leg of power). Otherwise, I followed the hardware setup as shown in the Wiki.

Any suggestions?

Thanks!

Saur0o0n, 2025-01-10:
Hi,
Like I wrote in the readme, I've just made 1.5 to compile with current esp framework, it's not very well tested yet (use release 1.3 - that one works definitely - but requires older esp ide). What does the serial output says? Does it produce any "guru meditation" dump?

jvswan, 2025-01-19:
Thank you! I hadn't picked up on the issue with v1.5. I rolled back to v1.3 and ESP v2.0.14 and got it working after fixing a few Asyncwebparameter calls. Unfortunately, I don't know what the serial output was. :-(. I'm not an Arduino power user... I'm not sure how to access logs on the board yet.

Again, thank you for getting me past the restart loop issue! Now I need to figure out why the web interface tells me "File not found" when I load the index.html page on 192.168.10.1. Ugh... It's always something!

Saur0o0n, 2025-01-24:
Hi, I'm not saying this is the case (but esp framework had done few braking changes in the past - so I wouldn't be surprised) - but perhaps try ESP closer to  2.0.4 or something (I remember this was the version the tests where done). 
I'll find the time to check the code with recent update I've made - but not yet :)

jvswan, 2025-03-09:
Thanks @Saur0o0n for your help. I've learned a few things... PIDKiln v1.3 installed ok. I went and installed older commits for the ESP32 framework and all the libraries. I downloaded commits that were current in January, 2022. That seemed to get all the pieces right, so the software started up ok. However, I ran into the restarting loop again once I got the data files uploaded. It took me a while to track down where the issue is. 

First, I found that the ESP32 Sketch Data Upload tool only works with Arduino IDE 1.x.x. I was using the most recent 2.x.x version... Once I installed the older IDE, I was able to upload the data files and didn't get the "File not found" error. 

Unfortunately, when I uploaded the data files the restart loop started up again. After flashing and uploading data files one at a time I discovered that the program files in the "programs" folder cause PIDKiln to go wonky. When I upload a program .txt file in the programs folder and restart PIDKiln, I cannot get the web pages to load and the LCD display shows PIDKiln is in a restart loop (keeps restarting over and over). This happens when I upload via the Sketch Data Upload tool in the IDE and also when I upload a program file in the WWW interface. I was using the test programs provided with the package, though. I haven't yet tried to upload a custom program yet, but I doubt it will behave differently. Those files are pretty simple.

Any suggestions? I can't test the relays until I can get a program file loaded. I'm sure it is something simple... Any help you can provide would be greatly appreciated!

jvswan, 2025-03-09:
BTW, here are the versions I'm using, in case it helps:

- PIDKiln v1.3
- Arduino ESP32 v2.0.2
- ESP32FS Plugin v1.0
- PID v1.2.1
- ESPAsyncWebServer v1.2.3
- AsyncTCP v1.2.2
- Syslog 2.0.0
- Max31855 v1.2.2
- U8G2 v2.35.30 (I didn't think it would matter if this library was newer than PIDKiln)

Saur0o0n, 2025-03-10:
Hi, so first of all you should get some serial debug output - it very verbose and there should be some feedback where (and probably why) it fails. Also if crushed, you will get memory/calls map to debug exactly where it failes.

Currently the SPIFFS is consider deprecated, so there will be no tools to upload it with Arduino (I guess) - you can still use command lines. Rewrite to LitteFS would require much testing, since there is quite a lot of quirks in PidKiln to overcome some SPIFFS issues.
Perhaps the best way to start would be to install binaries I provided (see wiki) but they are compiled for Wrover as I remember (not Wroom).

jvswan, 2025-03-12:
Thank you for the info. I’m not sure how to get serial debug feedback. Do I need to turn debugging back on in the sketch? Where would I find the debugging log? I’m sure it is in the documentation somewhere. I’ll find it…

The version of the IDE I’m using will load the files in the data folder. I have it set to flash the chip before every upload, so I don’t have old versions of files hiding in there. I don’t need to use the command line to load anything, thankfully, unless I’m misunderstanding something. 

Currently everything in the data folder is loaded except the program files because those for some reason cause the restart loop. I’m sure it isn’t the files specifically, but there is something that happens around the time PIDKiln wants to connect to WiFi that a restart is triggered, but only when there’s a program file in the programs folder. If the folder is empty, no restart. 

So, regarding the binaries, are you suggesting I reset the ESP32 board and reinstall using the binaries instead of using the IDE? I’ll give it a shot.

First I’ll see if I can learn more about debugging and post a log with the restart failure going on.

jvswan, 2025-03-17:
@Saur0o0n Thanks for helping me with this project. I'm not sure why I cannot seem to make progress on this issue. I have not been able to retrieve any logs. I have serial console output set to "yes" but when I go to the server IP using the indicated port (192.168.1.2:514) I get a blank "About" page. I'm at my wit's end with this. Any other suggestions for getting log data? 

I wasn't able to use the binary to install PIDKiln because I couldn't figure out how to get the OTA to work. I tried uploading the files after adding my wifi info to OTA_Install.ino, but after the card reboots there is no pidkilin_AP network to join. I'm sure I'm doing something wrong, but I can't figure out what to do from the wiki page.

Any additional info would be great. I'm completely stumped. About to find a C++ programmer at work and pay them to go through the code and explain to me what's going on.