Saur0o0n/PIDKiln #51: User abort 16 seconds after program start

terrin2k, 2025-01-25:
This is most likely user error, but any time I start a program, the EMRs energize, but not the SSRs, the the program spontaneously aborts after just a few seconds.

I've verified that there is no voltage on pins 19 and 22 when the program is running. 

I do not have the kiln heater coils connected to the SSRs while I'm testing things out.

Here's one example of the log output:

Program name: test_up_down.txt
Program desc: Program to test up and down temp. change
Started at: 2025-01-25 15:30:39
Possible end at: 2025-01-25 15:50:39
PID values. Kp:20.00 Ki:0.20 Kd:0.10
Start temperature: 14.5C
CSV filename: 2025-01-25 15:50:39
-=-=-= Starting program =-=-=-=-
Program ended at: 2025-01-25 15:30:55
End temperature: 14.5C
Program aborted with error: 9

Any idea what I might be doing wrong?

Thanks!

~ D

Saur0o0n, 2025-01-25:
Hi, you have information in log - error 9.

/* 
**  Program errors:
*/
typedef enum {
  PR_ERR_FILE_LOAD,       // failed to load file
  PR_ERR_TOO_LONG_LINE,   // program line too long (there is error probably in the line - it should be max. 1111:1111:1111 - so 14 chars, if there where more PIDKiln will throw error without checking why
  PR_ERR_BAD_CHAR,        // not allowed character in program (only allowed characters are numbers and separator ":")
  PR_ERR_TOO_HOT,         // exceeded max temperature defined in MAX_Temp
  PR_ERR_TOO_COLD,        // temperature redout below MIN_Temp
  PR_ERR_TOO_HOT_HOUSING, // housing temperature exceeded
  PR_ERR_MAX31A_NC,       // MAX31855 A not connected
  PR_ERR_MAX31A_INT_ERR,  // failed to read MAX31855 internal temperature on kiln
  PR_ERR_MAX31A_KPROBE,   // failed to read K-probe temperature on kiln
  PR_ERR_MAX31B_NC,       // MAX31855 B not connected
  PR_ERR_MAX31B_INT_ERR,  // failed to read MAX31855 internal temperature on case
  PR_ERR_MAX31B_KPROBE,   // failed to read K-probe temperature on case
  PR_ERR_USER_ABORT,      // user aborted
  PR_ERR_end
} PROGRAM_ERROR_STATE;

terrin2k, 2025-01-25:
I must have miscounted. I thought error 9 was "user abort". :)

Thanks! 

~ D

terrin2k, 2025-02-02:
Replaced the case thermocouple and it is much happier now! Thanks!

closed 2025-02-02 by terrin2k