Saur0o0n/PIDKiln #52: EMRs activate but SSRs do not

terrin2k, 2025-02-03:
I have two SSRs. One is connected to pin 19 and the other to pin 22, per the instructions. I've verified that there are no bad connections. When the program starts, the EMRs click on (and LEDs on the EMR boards illuminate). However, the voltage at pins 19 and 22 stays at zero and so the SSRs never activate.

Any idea what boneheaded mistake I've made this time? :)

Regards, 

~ D

terrin2k, 2025-02-23:
Circling back to this. Sorry for the delay.. been dealing with illness here.

Anyway, I made a simple program to set up and drive pins 19 and 22 high and it works. Both my SSRs energize when I run this program.

When I run PIDKiln and start a program to heat the kiln, it energizes the EMRs but not the SSRs. Here's my current configuration.. perhaps I have something wrong here?

Thanks for your help and an excellent program!

~ D

WiFi_SSID = xxxx
WiFi_Password = xxxx
WiFi_Mode = 1
WiFi_Retry_cnt = 9
HTTP_Local_JS = 1
Auth_Username = admin
Auth_Password = xxxxxx
NTP_Server1 = 0.pl.pool.ntp.org
NTP_Server2 = 1.pl.pool.ntp.org
NTP_Server3 = 2.pl.pool.ntp.org
GMT_Offset_sec = -21600
Daylight_Offset_sec = 3600
Initial_Date = 2024-01-01
Initial_Time = 11:00:00
PID_Window = 5000
PID_Kp = 20.00
PID_Ki = 0.20
PID_Kd = 0.10
PID_POE = 0
PID_Temp_Threshold = 5   <-- I've tried experimenting with this with no luck so far.
LOG_Window = 30
LOG_Files_Limit = 40
MIN_Temperature = 0
MAX_Temperature = 1350
MAX_Housing_Temperature = 130
Thermal_Runaway = 0
Alarm_Timeout = 5
MAX31855_Error_Grace_Count = 10
DBG_Serial = 1
DBG_Syslog = 0
DBG_Syslog_Srv = 192.168.1.2
DBG_Syslog_Port = 514

Saur0o0n, 2025-03-10:
Hi,
Please take a look at serial debug output - perhaps it doesn't think it's require to start heating. The EMR relay is started with program start, SSRs are started only when heating is required.

terrin2k, 2025-04-26:
I finally got back around to trying to get this to work.

A couple of updates:
* I constructed a transistor stage to ensure my GPIO can drive the SSR inputs. I've tested this and it works great.
* I rebuilt the code with debugging enabled and captured the attached output while loading and running the Slow Cone 04 program from another user on this message board.

Can you make heads or tails of it?

Best regards,

~ D

[pidkiln-debug.txt](https://github.com/user-attachments/files/19924926/pidkiln-debug.txt)

Saur0o0n, 2025-04-26:
Hi,
So there is few things strange/wrong. Easy one is the threshold window - you have a lot of messages like:
```
[PRG] Temperature in TEMP_THRESHOLD. Kiln_temp:21.4 Set_temp:26.5 Window:5
```
So if current temp is not more different from set_temp then threshold window - do nothing. So it will do nothing.

Another thing. It will only enable heating (ssr) if this part is true
```
if (pid_out*PID_WINDOW_DIVIDER > now - windowStartTime) Enable_SSR();
```
In logs you can read those values here:
```
[PRG] Pid_out RAW:58.46 Pid_out:58.46 Now-window:-3343913 WindowSize:5000 Prg_state:6
```
Here pid_out - is the same as "pid_out*PID_WINDOW_DIVIDER" - and "Now-window" from logs is the "now - windowStartTime".
So if "now-window" is negative - it should trigger heating all the time because 58.46>-3343913

So I would assume there is something wrong with your physical connection on the board if the SSR is not enabled.
There is no any debug in enable_ssr function, but you can add it for your purpose in PIDKiln_addons.ino here

```
// Simple functions to enable/disable SSR - for clarity, everything is separate
//
void Enable_SSR(){
  if(!SSR_On){
    digitalWrite(SSR1_RELAY_PIN, HIGH);
    DBG dbgLog(LOG_ERR,"[ADDONS] Enabling SSR1\n");
#ifdef SSR2_RELAY_PIN
    digitalWrite(SSR2_RELAY_PIN, HIGH);
    DBG dbgLog(LOG_ERR,"[ADDONS] Enabling SSR2\n");
#endif
    SSR_On=true;
  }
}
```