Saur0o0n/PIDKiln #53: How do you upload the filesystem when ESP32FS does not support current versions of the arduino IDE?

digiexchris, 2025-04-17:
I haven't been able to successfully compile PIDKiln with the legacy IDE, but I can compile and upload with the current 2.x version of the arduino ide. Could you please help find a way to upload the filesystem with the newer version?

Saur0o0n, 2025-04-19:
Hi,
You should be able to do this with cli tool mkspiffs - haven't tested it. The quick guide is here - https://coolplaydev.com/esp32-spiffs-file-uploads-without-arduino-ide. Please report back if you manage to use it - it may be useful for other too.

digiexchris, 2025-04-22:
hey thanks for this, I was able to do it this way
./mkspiffs -c /path/to/PIDKiln/data -b 4096 -p 256 -s 0x160000  spiffs.bin
esptool.py  --chip esp32 write_flash -z 0x290000 spiffs.bin

I have the esp32 toolchain setup for espidf work for another project, so esptool.py was available through that, though I'm pretty sure the arduino esp32 board libraries include it. I needed to build mkspiffs myself, but I suspect there are versions available to download out there